ros-industrial/industrial_ci #583: Private repository access is broken

stertingen, 2021-01-18:
Currently, the private GitLab respository access (as documented in https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#gitlab-ci-access-to-private-repositories) seems to be broken. 

```
Could not clone repository 'ssh://git@<...>.git': Cloning into '.'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
```

Until a few days ago, SSH was available automatically when SSH_PRIVATE_KEY was set.

ipa-mdl, 2021-01-18:
Thanks for reporting!
This might be a regression of #572

Could share (here or by mail or web meeting) more details about your job config?
There might be as well an error line at the front of the log.

I will review my changes around SSH handling in the mean-time.

stertingen, 2021-01-18:
**.gitlab-ci.yml:**
```yaml
# The following block is needed for the shared Docker-based runner
# For local runners you might want to enable the overlay driver:
# https://docs.gitlab.com/ce/ci/docker/using_docker_build.html#using-the-overlayfs-driver

image: docker:git # docker and git clients

# The docker runner does not expose /tmp to the docker-in-docker service
# This config ensures that the temp folder is located inside the project directory (e.g. for prerelease tests or SSH agent forwarding)
variables:
  CATKIN_LINT: "pedantic"
  CCACHE_DIR: "${CI_PROJECT_DIR}/ccache"
  DOCKER_RUN_OPTS: "-v ${CI_PROJECT_DIR}/.build:/root/target_ws/build"
  TMPDIR: "${CI_PROJECT_DIR}.tmp"
  UPSTREAM_WORKSPACE: ".rosinstall"

cache:
  key: "${CI_JOB_NAME}" # https://docs.gitlab.com/ee/ci/caching/#sharing-caches-across-different-branches
  paths:
    - ccache

# enable docker-in-docker
services:
  - docker:19.03.5-dind

before_script:
  - apk add --update bash coreutils tar grep # install industrial_ci dependencies
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master

default:
  artifacts:
    when: always
    reports:
      junit: ${CI_PROJECT_DIR}/.build/**/test_results/**/*.xml

# setup the actual tests

format:
  script: .industrial_ci/gitlab.sh
  variables:
    CLANG_FORMAT_CHECK: "file"
    CLANG_FORMAT_VERSION: "8"
    ROS_DISTRO: "noetic"

kinetic:
  script: .industrial_ci/gitlab.sh
  variables:
    ROS_DISTRO: "kinetic"

melodic:
  script: .industrial_ci/gitlab.sh
  variables:
    ROS_DISTRO: "melodic"

noetic:
  script: .industrial_ci/gitlab.sh
  variables:
    ROS_DISTRO: "noetic"
```

Also, `SSH_PRIVATE_KEY` and `SSH_SERVER_HOSTKEYS` are set according to the docs, the public key is added as deploy key to upstream repositories.

ipa-mdl, 2021-01-18:
Thanks! Nothing suspicious so far..
I will to give it a try myself later.

ipa-mdl, 2021-01-18:
@stertingen: I think I found the issue, please test #584

stertingen, 2021-01-18:
> @stertingen: I think I found the issue, please test #584

I've hit the pull rate limit on the Docker Hub, I'm trying again tomorrow.

stertingen, 2021-01-18:
Gave it another try. The PR does seem to fix the issue. Thank you!

PR #584 (2021-01-18): DOCKER_COMMIT_CREDENTIALS must not default to false
closes #583
closed 2021-01-18 by ipa-mdl