ros-industrial/industrial_ci #586: Startup failure error reported on my ros repo

meitiever, 2021-01-21:
I added the template file to the code base - .github/workflows/ci.yml
but the action tab reports Startup failure and not more.
Any idea?

ipa-mdl, 2021-01-21:
> Any idea?

No..
But it might help if you would clarify which "template file" you added and what the "Startup failure" looks like :)

meitiever, 2021-01-21:
name: CI

# This determines when this workflow is run
on: [push, pull_request] # on all pushes and PRs

# or more fine-grained
# on:
#   # When master or melodic-devel branch is pushed to
#   push:
#     branches: [ master, melodic-devel ]
#   # When there is a pull request against master
#   pull_request:
#     branches: [ master ]

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: melodic, ROS_REPO: testing}
          - {ROS_DISTRO: melodic, ROS_REPO: main}
    env:
      CCACHE_DIR: /github/home/.ccache # Directory for ccache (and how we enable ccache in industrial_ci)
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
      # This step will fetch/store the directory used by ccache before/after the ci run
      - uses: actions/cache@v2
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      # Run industrial_ci
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}

ipa-mdl, 2021-01-21:
I just gave it a try and copied  [industrial_ci template](https://github.com/ros-industrial/industrial_ci/blob/master/doc/industrial_ci_action.yml) to the  [ros_canopen repository](https://github.com/ipa-mdl/ros_canopen/commit/17697b98c5e45864792f9682d8074d8ea165be3c).
It works as expected: https://github.com/ipa-mdl/ros_canopen/runs/1741721720

[image]

Please double check that your workflow file is formatted correctly.

meitiever, 2021-01-31:
thx. the process is working now.
and now, I meet the following error:
Function 'build_target_workspace' returned with code '1' after 0 min 23 sec

meitiever, 2021-01-31:
I have tested some ros packages
and the following package will cause the error:
this is the cmakelist.txt and no source code:
cmake_minimum_required(VERSION 3.5.0)
project(gflags_catkin)

find_package(catkin_simple REQUIRED)
catkin_simple()

include(ExternalProject)

file(MAKE_DIRECTORY ${CATKIN_DEVEL_PREFIX}/include)

ExternalProject_Add(
  gflags_src
  URL https://github.com/gflags/gflags/archive/v2.1.2.zip
  URL_MD5 5cb0a1b38740ed596edb7f86cd5b3bd8
  UPDATE_COMMAND ""
  CONFIGURE_COMMAND cd ../gflags_src &&
     cmake . 
       -DCMAKE_INSTALL_PREFIX:PATH=${CATKIN_DEVEL_PREFIX}
       -DBUILD_SHARED_LIBS:BOOL=true
       -DGFLAGS_NAMESPACE:STRING=google
       -DCMAKE_BUILD_TYPE:STRING=Release
  BUILD_COMMAND cd ../gflags_src && make -j 8
  INSTALL_COMMAND cd ../gflags_src && make install -j 8
)

cs_add_library(${PROJECT_NAME} src/dependency_tracker.cc)
add_dependencies(${PROJECT_NAME} gflags_src)
target_link_libraries(${PROJECT_NAME} 
  ${CATKIN_DEVEL_PREFIX}/lib/libgflags${CMAKE_SHARED_LIBRARY_SUFFIX})

cs_install()

cs_export(INCLUDE_DIRS ${CATKIN_DEVEL_PREFIX}/include
  CFG_EXTRAS gflags-extras.cmake)

ipa-mdl, 2021-02-01:
It is good to hear that the CI job is running now.
What did you change?

There is no general advice on how to fix your build issues.
However, I can recommend the [catkin documentation](http://docs.ros.org/en/melodic/api/catkin/html/howto/format2/index.html).
Please note that `industrial_ci` does not support the devel space anymore, so you might have to add install tags etc. (as described in the link) to make the build pass.

I willl close this issue now, because there is nothing we can do about it in `industrial_ci`.

PS: Please format your [posts properly](https://docs.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax#quoting-text)..

closed 2021-02-01 by ipa-mdl