ros-industrial/industrial_ci #599: github actions CI running tests twice

k2shah, 2021-01-29:
Hello,
 We recently setup our a github action to build and test our ros code. However it looks like the tests are being run twice. When `Starting function 'run_target_test'` is run it shows `Summary: 8 tests, 0 errors, 0 failures, 0 skipped` 8 tests being run but we only have 4 tests. 
 
 Our test are here
 https://github.com/StanfordMSL/msl_quad/blob/test-setup/mslquad/test/test_hello.cpp 
 they are just sample tests to work out all the CI setup.  When these tests are run locally (with both `catkin_make` and `catkin run_tests` there are only 4 tests.
 
 I'm not sure if this is a CI setup issue on our end of if we are running the tests incorrectly. Any help would be appreciated!

ipa-mdl, 2021-01-29:
I just ran some local tests with your repository and cannot tell why 8 tests get reported..

catkin_test_results lists
> build/mslquad/test_results/mslquad/gtest-mslquad-test.xml: 8 tests

And this file only lists 4 tests:
```xml
<?xml version="1.0" encoding="UTF-8"?>                                                                                                                                      <testsuites tests="4" failures="0" disabled="0" errors="0" timestamp="2021-01-29T07:51:35" time="0" name="AllTests">
  <testsuite name="TestSuite" tests="4" failures="0" disabled="0" errors="0" time="0">
     <testcase name="testHello" status="run" time="0" classname="TestSuite" />
     <testcase name="testSucceed" status="run" time="0" classname="TestSuite" />
     <testcase name="testAssert" status="run" time="0" classname="TestSuite" />
     <testcase name="testFloat" status="run" time="0" classname="TestSuite" />
   </testsuite>
</testsuites>

ipa-mdl, 2021-01-29:
If I feed this into `catkin_test_results` (noetic) on my computer, then it reports 8 tests as well.
It might be a bug in [catkin](https://github.com/ros/catkin).

ipa-mdl, 2021-01-29:
This seems to be a regression of https://github.com/ros/catkin/pull/1033.
I will close this issue here.

closed 2021-01-29 by ipa-mdl