ros-industrial/industrial_ci #607: Issue with GitHub Actions / image

destogl, 2021-02-01:
There is an issue with updating the `msodbcsql17` package when using GitHub Actions and the ubuntu-latest image.

You can see an example of the build log [here](https://github.com/ros-controls/ros2_control/runs/1805158937).

Has anyone a solution, or should I be patient?

gavanderhoorn, 2021-02-01:
I'm not sure more patience would help:

```
Preparing to unpack .../29-msodbcsql17_17.7.1.1-1_amd64.deb ...
ERROR: The EULA was not accepted. Installation aborted.
dpkg: error processing archive /tmp/apt-dpkg-install-HnuQT7/29-msodbcsql17_17.7.1.1-1_amd64.deb (--unpack):
 new msodbcsql17 package pre-installation script subprocess returned error exit status 1
```

it seems that package requires an EULA to be accepted by the user, and that's probably not going to happen in a CI run, run in a Docker container.

ipa-mdl, 2021-02-01:
> It seems that package requires an EULA to be accepted by the user

`DOCKER_RUN_OPTS=-e ACCEPT_EULA=Y` [should work](https://docs.microsoft.com/en-us/sql/connect/odbc/linux-mac/installing-the-microsoft-odbc-driver-for-sql-server?view=sql-server-2017) then.

ipa-mdl, 2021-02-01:
However, in this case, just remove the [apt update line](https://github.com/ros-controls/ros2_control/blob/eb59ce0c445eed88d6ab42296e5f2854496b5229/.github/workflows/ci.yml#L21).
It is not needed at all.

closed 2021-02-02 by destogl