ros-industrial/industrial_ci #608: Recurrent 'failed to fetch' at 'setup_upstream_workspace' step in GHA

gavanderhoorn, 2021-02-02:
In [this build](https://github.com/ros-industrial/abb_robot_driver/runs/1812458779?check_suite_focus=true), which is a scheduled build, it often happens (well, sufficiently often) that `apt-get` is unable to download certain `.deb`s.

It's not really the same `.deb`s necessarily, but there have been a few runs now where it's unable to install all required archives and then quits.

This seems similar to https://github.com/actions/virtual-environments/issues/675 (although that's with a different package host).

This issue is not a this-is-broken-please-fix kind of issue, just wanted to see whether others have encountered this as well and how they've dealt with it.

gavanderhoorn, 2021-02-02:
Oh, forgot to mention: rerunning the workflow so far has always resulted in a successful build.

ipa-mdl, 2021-02-02:
> how they've dealt with it

So far I have just restarted failed builds..
Not sure if restarting the command itself would help (general connectivity issue?).

Do Github Actions use mirrors or proxies? 
If so, then  something like https://github.com/ros-industrial/industrial_ci/pull/424 might be an option.

gavanderhoorn, 2021-02-02:
> Not sure if restarting the command itself would help (general connectivity issue?).

you mean adding retries to `apt install`? I've read about ppl that did this, but apparently it just delays, it doesn't solve the problem.

> Do Github Actions use mirrors or proxies?

I don't know actually.

gavanderhoorn, 2021-02-11:
Perhaps related: today I encountered:

```
 E: Unable to locate package ros-melodic-catkin
```

in `catkin_tools_setup`.

Not sure what was going on there.

ipa-mdl, 2021-02-11:
>  E: Unable to locate package ros-melodic-catkin

The [new sync](https://discourse.ros.org/t/melodic-sync-job-failed/18901/2) fixed our builds [on Gitlab](https://gitlab.com/ipa-mdl/industrial_ci/-/jobs/1024113004).

gavanderhoorn, 2021-02-11:
Yes, same for me.

ipa-mdl, 2021-07-21:
@gavanderhoorn: can we close this?

closed 2021-09-16 by ipa-mdl