ros-industrial/industrial_ci #618: ici action fails with jq command not found

fmessmer, 2021-02-10:
for the following config:
```
name: CI

on:
  push:
  pull_request:
#  schedule:
#    - cron: "0 0 * * 0" # every Sunday at midnight

jobs:
  industrial_ci:
    if: github.event.pull_request.draft == false  # only run on non-draft PRs
    runs-on: self-hosted
    timeout-minutes: 60

    env:
      ADDITIONAL_DEBS: 'apt-utils curl dialog'
      CATKIN_LINT: pedantic
      CATKIN_LINT_ARGS: '--ignore description_boilerplate'
      CCACHE_DIR: /github/home/.ccache  # Directory for ccache (and how we enable ccache in industrial_ci)
      CMAKE_ARGS: -DCMAKE_BUILD_TYPE=Release
      PARALLEL_TESTS: 1
      PYLINT_ARGS: '--output-format=parseable --errors-only'
      PYLINT_CHECK: true
      ROS_REPO: main
      SECRET_AVAILABLE: ${{secrets.GITHUBPAT}}
      UNDERLAY: '/root/git/care-o-bot/devel'  # workspace on docker image

    strategy:
      fail-fast: false
      matrix:
        include:
          - { ROS_DISTRO: kinetic,
              DOCKER_IMAGE: 'ghcr.io/mojin-robotics/mojin-ros:kinetic-full',
              TARGET_WORKSPACE: 'mojin_atf_kevin'
            }

    steps:
      - uses: actions/checkout@v2
        with:
          lfs: true
          ssh-known-hosts: ''
      - uses: 'ros-industrial/industrial_ci@master'
        with:
          config: ${{toJSON(matrix)}}
```

I get the following failure
```
Run ros-industrial/industrial_ci@master
  with:
    config: {
    "ROS_DISTRO": "kinetic",
    "DOCKER_IMAGE": "ghcr.io/mojin-robotics/mojin-ros:kinetic-full",
    "TARGET_WORKSPACE": "mojin_atf_kevin"
  }
  env:
    ADDITIONAL_DEBS: apt-utils curl dialog
    CATKIN_LINT: pedantic
    CATKIN_LINT_ARGS: --ignore description_boilerplate
    CCACHE_DIR: /github/home/.ccache
    CMAKE_ARGS: -DCMAKE_BUILD_TYPE=Release
    PARALLEL_TESTS: 1
    PYLINT_ARGS: --output-format=parseable --errors-only
    PYLINT_CHECK: true
    ROS_REPO: main
    SECRET_AVAILABLE: ***
    UNDERLAY: /root/git/care-o-bot/devel
/home/github-action-runner/actions-runner1/_work/_actions/ros-industrial/industrial_ci/master/.github/action.sh: Zeile 28: jq: Befehl nicht gefunden

Running test 'source_tests'
Please set ROS_DISTRO, OS_CODE_NAME or DOCKER_IMAGE.
Error: Process completed with exit code 1.
```

fmessmer, 2021-02-10:
this only happens on `self-hosted` runners....I think ICI should make sure `jq` is installed, right?

ipa-mdl, 2021-02-10:
>  ICI should make sure jq is installed, right?

At least it should complain about this missing dependency.
Not sure, if we should try to install it..

ipa-mdl, 2021-02-10:
> Please set ROS_DISTRO, OS_CODE_NAME or DOCKER_IMAGE.

At least #617 is working properly ;)

ipa-mdl, 2021-07-16:
Will close this for now.

closed 2021-07-16 by ipa-mdl