ros-industrial/industrial_ci #621: [pylint] segmentation fault

fmessmer, 2021-02-12:
also, mainly for documentation - I don't have sufficient insights yet
```
Starting function 'run_pylint'
No config file found, using default configuration
/usr/lib/python2.7/dist-packages/astroid/brain/brain_gi.py:136: PyGIWarning: Gdk was imported without specifying a version first. Use gi.require_version('Gdk', '2.0') before import to ensure that the right version gets loaded.
  __import__(m)
/home/fxm/projects/ci/install/share/industrial_ci/src/workspace.sh: line 282:  4438 Segmentation fault      (core dumped) ( { [ ! -e "$extend/setup.bash" ] || ici_source_setup "$extend"; } && cd "$path" && exec "$@" )
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'run_pylint' returned with code '139' after 0 min 24 sec
Tests failed exit code '139'
```

fmessmer, 2021-02-12:
seems to come from running pylint on a file that imports `gi`
e.g. https://github.com/ipa320/cob_command_tools/blob/08bb61ad0637d238c7d1a114e7db39f84a277904/cob_command_gui/src/knoeppkes.py#L19

this import writes on stderr and apparently this makes https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/workspace.sh#L282 crash which is called from https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/tests/source_tests.sh#L117

ipa-mdl, 2021-02-12:
Why not follow their recommendation?
> Use gi.require_version('Gdk', '2.0') before import to ensure that the right version gets loaded.

fmessmer, 2021-02-12:
while `gi.require_version('Gdk', '2.0')` gets rid of the warning above, apparently `pylint` still segfaults on [`knoeppkes.py`](https://github.com/ipa320/cob_command_tools/blob/indigo_dev/cob_command_gui/src/knoeppkes.py) - even running `pylint` on this file locally

I've added a local `skip-file` in https://github.com/ipa320/cob_command_tools/pull/296

so this doesn't seem to be an issue of the pylint feature in industrial_ci

ipa-mdl, 2021-02-12:
>do you think we still need to handle a possible segfault in industrial_ci?

The test fails.. what else should be taken care of?

ipa-mdl, 2021-02-12:
> so this doesn't seem to be an issue of the pylint feature in industrial_ci

It would be good to know, why it segfaults..

fmessmer, 2021-02-12:
> > so this doesn't seem to be an issue of the pylint feature in industrial_ci
> 
> It would be good to know, why it segfaults..

yes, still havent identified the real reason yet...still investigating...it rather seems not to be related to ici doing something wrong...

fmessmer, 2021-02-15:
this is no issue related to ICI's pylint check

closed 2021-02-15 by fmessmer