ros-industrial/industrial_ci #623: ROS1 noetic support for #!/usr/bin/env python

stwirth, 2021-02-12:
I just tried to add a noetic build to flir_camera_driver: https://github.com/ros-drivers/flir_camera_driver/pull/67
That package uses a download script written in python to install a dependency.
The [Travis CI job](https://travis-ci.org/github/ros-drivers/flir_camera_driver/jobs/758689854) reports
```
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'build_target_workspace'
Starting >>> spinnaker_camera_driver
--- stderr: spinnaker_camera_driver
/usr/bin/env: ‘python’: No such file or directory
```
and then fails due to a missing include as the dependency was not downloaded.
I am assuming that the problem is that the default python executable on 20.04 is called `python3` and not `python`.
Maybe the build environment should have `python-is-python3` installed?

gavanderhoorn, 2021-02-12:
isn't it actually the responsibility of the maintainer to change the shebang to `python3` explicitly, to signal that the script supports Python 3 and doesn't work with any other interpreter?

`python-is-python3` is a work-around, not a fix.

stwirth, 2021-02-12:
The script works with python2 as well. But I agree, the script should be ported to python3 instead.

closed 2021-02-12 by stwirth

gavanderhoorn, 2021-02-12:
You could use one of the hooks, or perhaps `ADDITIONAL_DEBS` to install `python-is-python3` at the appropriate time.

ipa-mdl, 2021-02-12:
Normally, you should use `catkin_install_python` to avoid exactly this problem.
http://docs.ros.org/en/noetic/api/catkin/html/howto/format2/installing_python.html

gavanderhoorn, 2021-02-12:
If it's a Catkin package, yes.

You could use that.

stwirth, 2021-02-12:
The script in question runs during configuration time of cmake, so I would need to move it to another package. I made it use `#!/usr/bin/env python3` which works fine.

stwirth, 2021-02-12:
(Using `ExternalProject_Add` in the `CMakeLists.txt` would probably be even better.)