ros-industrial/industrial_ci #624: rosenv does not correctly consider CMAKE_PREFIX_PATH in AFTER_SCRIPT

fmessmer, 2021-02-15:
I made the following observation:
- using Github Actions (does NOT happen on travis)
- for `noetic` :point_right: default builder is `colcon` (does NOT happen on `kinetic` :point_right: default builder `catkin-tools`)
- using `UNDERLAY` option with custom docker image
- `AFTER_SCRIPT: 'rosenv bash .travis.xacro_test.sh'` 
  (see https://github.com/ipa320/cob_common/blob/kinetic_dev/.travis.yml#L12)

for some reason in GHA noetic job, the test fails because the resources for the xacro test cannot be found, although they should be findable in the `$CMAKE_PREFIX_PATH`

```
   Starting function 'after_script'
  Testing /home/runner/work/mojin_common/mojin_common/mojin_description/urdf/base_theta/base_theta.urdf.xacro
  Testing /home/runner/work/mojin_common/mojin_common/mojin_description/urdf/mrl_base/base.gazebo.xacro
  Testing /home/runner/work/mojin_common/mojin_common/mojin_description/urdf/mrl_base/base.urdf.xacro
  resource not found: cob_description
  ROS path [0]=/opt/ros/noetic/share/ros
  ROS path [1]=/root/target_ws/install/mojin_common/share
  ROS path [2]=/root/target_ws/install/mojin_utils/share
  ROS path [3]=/root/target_ws/install/mojin_msgs/share
  ROS path [4]=/root/target_ws/install/mojin_description/share
  ROS path [5]=/opt/ros/noetic/share
```

the missing resource `cob_description` is part of the `robot_ws` - see `CMAKE_PREFIX_PATH` below

I'm not sure whether my `AFTER_SCRIPT` is not colcon-ready or whether `rosenv` is doing something incorrect

EDIT:
here is the output of `AFTER_SCRIPT: 'rosenv && echo "$CMAKE_PREFIX_PATH" && echo "$UNDERLAY"'`
- for kinetic:
   ```
  Starting function 'after_script'
  /root/target_ws/install:/root/git/app_ws/install:/root/git/rosjava_ws/install:/root/git/mojin_ws/install:/root/git/nav_ws/install:/root/git/robot_ws/install:/opt/ros/kinetic:/root/git/care-o-bot/devel
  /root/git/care-o-bot/devel
   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
   Function 'after_script' returned with code '0' after 0 min 0 sec
   ```
- for noetic:
   ```
  Starting function 'after_script'
  /root/target_ws/install/mojin_utils:/root/target_ws/install/mojin_msgs:/root/target_ws/install/mojin_description:/root/git/app_ws/install:/root/git/rosjava_ws/install:/root/git/mojin_ws/install:/root/git/nav_ws/install:/root/git/robot_ws/install:/opt/ros/noetic
  /root/git/care-o-bot/devel
   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
   Function 'after_script' returned with code '0' after 0 min 0 sec
   ```

ipa-mdl, 2021-02-15:
Are you building `UNDERLAY` with colcon as well? Then you should use `--merge-install`. Otherwise, it cannot be extended.

ipa-mdl, 2021-02-15:
I doubt that `xacro` would use `CMAKE_PREFIX_PATH`. I guess it relies on `rospack` and there needs `ROS_PACKAGE_PATH`.

fmessmer, 2021-02-15:
no, the `UNDERLAY` workspace is built with `catkin-tools` (`catkin config --install`)
I'll also debug the `ROS_PACKAGE_PATH`...

fmessmer, 2021-02-15:
when I reproduce this locally, i.e. 
- have a catkin-tools workspace chain (using install space)
- source the topmost `setup.bash` of the catkin-tools chain
- CMAKE_PREFIX_PATH and ROS_PACKAGE_PATH are as expected
- create a colcon workspace on top
- colcon build
- `source install/local_setup.bash`
- CMAKE_PREFIX_PATH and ROS_PACKAGE_PATH are still as expected

what's different in ICI? or the `rosenv` function?

ipa-mdl, 2021-02-15:
> source install/local_setup.bash

`industrial_ci` sources `setup.bash`

ipa-mdl, 2021-02-15:
And `setup.bash` [must be present](https://github.com/ros-industrial/industrial_ci/blob/98b5f6756941b74a82bae88cd254b0fbc0cb3ca8/industrial_ci/src/util.sh#L67)!

fmessmer, 2021-02-15:
hmm...when I `unset CMAKE_PREFIX_PATH` and `ROS_PACKAGE_PATH` before sourcing either `colcon_ws/install/setup.bash` OR `colcon_ws/install/local_setup.bash` (`setup.bash` uses `local_setup.bash), `CMAKE_PREFIX_PATH` and `ROS_PACKAGE_PATH` only include the `share` folders of the `colcon_ws` - even though the underlay workspace was correctly sourced before

but in my CI build, there is still the `/opt/ros/noetic/share` included....so it seems not to consider what I set as `UNDERLAY` in the `main.yml`, but defaults to `/opt/ros/$ROS_DISTRO`

could that be it?

ipa-mdl, 2021-02-15:
Have you tried building everything with the same builder?
As far as I remember, mixing build tools might have some issues, especially because `catkin-tools` has been supported officially...

fmessmer, 2021-02-15:
obviously, sourcing a colcon workspace over a catkin-tools workspace configures the CMAKE_PREFIX_PATH correctly, but the ROS_PACKAGE_PATH is incorrect
using `BUILDER=catkin_tools` in the noetic job works fine...

ipa-mdl, 2021-02-15:
> but the ROS_PACKAGE_PATH is incorrect

`CMAKE_PREFIX_PATH` is managed by colcon. `ROS_PACKAGE_PATH` is setup from `catkin` (not catkin_tools), but needs the enviroment hooks from the [ros_environment](https://github.com/ros/ros_environment) package.

ipa-mdl, 2021-02-15:
I ran some tests and I don't think there is anything wrong with `rosenv`, but there is an issue with chaining these workspaces..

ipa-mdl, 2021-02-15:
> but there is an issue with chaining these workspaces..

`local_setup.bash` does not seem to work properly, as it does not update `ROS_PACKAGE_PATH`

ipa-mdl, 2021-02-15:
> local_setup.bash does not seem to work properly, as it does not update ROS_PACKAGE_PATH

The `local_setup.bash` generated by `catkin` works differently from the one generated by `colcon`.
The problem is that `colcon`'s `setup.bash` just calls all `local_setup.bash` files in order, but only `/opt/ros/$ROS_DISTRO/local_setup.bash` and the colcon-generated `local_setup.bash`files contribute to `ROS_PACKAGE_PATH`.

ipa-mdl, 2021-02-15:
This is not only  problem with `rosenv` but as well for `ici_exec_in_workspace`.
One work-around would be to always source `UNDERLAY` and then one of the others.

PR #630 (2021-02-25): (Re)populate ROS_PACKAGE_PATH from all underlays
Fixes #624

@fmessmer: please test
closed 2021-02-25 by ipa-mdl