ros-industrial/industrial_ci #629: catkin_tools for ROS noetic

stertingen, 2021-02-24:
> EDIT: As workaround, we could set catkin_tools as default for noetic. Works fine for me.

@stertingen: I tried it with a dummy package and it did not work for me.

_Originally posted by @ipa-mdl in https://github.com/ros-industrial/industrial_ci/issues/475#issuecomment-785158785_

stertingen, 2021-02-24:
What did you try?

ipa-mdl, 2021-02-24:
> What did you try?

I tried a cmake-only package, which did not install package.xml ;)
See https://github.com/ros-industrial/industrial_ci/issues/475#issuecomment-785182656

`catkin_tools` itself works (since #625).

wxmerkt, 2021-06-07:
Hi,
Has this been resolved? E.g. I have cmake-only packages that work fine if I explicitly set the `BUILDER=catkin_tools` for Noetic but fail under colcon in Noetic (which is the default -- they work fine with colcon for ROS2/Foxy etc.). Would a resolution be to default `BUILDER=catkin_tools` for Noetic?

ipa-mdl, 2021-06-07:
> Would a resolution be to default BUILDER=catkin_tools for Noetic?

I would like to avoid this.
`catkin_tooks` is not officially recommended as a build tool.
If possible, I would rather make `colcon` the default for all non-EOL ROS1 distros.

> E.g. I have cmake-only packages that work fine if I explicitly set the BUILDER=catkin_tools for Noetic but fail under colcon in Noetic

What exactly does not work with colcon?
Did you register everything to the index?

wxmerkt, 2021-06-07:
The failures have to do with 

i) pure CMake projects in the `upstream_ws` which cannot be found by the `target_ws`. Those packages have `package.xml` that are being installed. They are not registered to an (ament?) index. What steps would I have to do to register them beyond the package.xml in a ROS1 context?

ii) tests are not being run, i.e. a ctest target exists (`make test`) but `run_tests` does not appear to trigger. This is likely a configuration issue on our side though or an unsupported use case.

ipa-mdl, 2021-06-07:
> They are not registered to an (ament?) index. 

My bad! I got confused..

> pure CMake projects in the upstream_ws which cannot be found by the target_ws

In which step? rosdep? cmake?

> tests are not being run, i.e. a ctest target exists (make test) but run_tests does not appear to trigger. This is likely a configuration issue on our side though or an unsupported use case.

Tests in `target_ws` should be run, but of cause not for `upstream_ws`.

---

Please have a look at [soss](https://github.com/ipa-mdl/soss_v2/blob/master/packages/core/CMakeLists.txt), which is pure CMake and works fine in an upstream workspace.

Some of these packages even contain CTest-based tests, which get run if they are part of the target (or downstream) workspace.

wxmerkt, 2021-06-07:
> In which step? rosdep? cmake?

This is with `rosdep`. When built with `catkin_tools`, `rosdep` is able to find pure CMake projects from the `upstream_ws` that install a `package.xml`. When built with `colcon`, `rosdep` outputs `cannot locate rosdep definition for [package]`. Do I need to do anything else to register the package?

> Tests in target_ws should be run, but of cause not for upstream_ws.

Correct, the issue was in target_ws. I noticed this was due to the way the targets were created: I.e., a `make test` target existed but not a `run_tests` target which is called by colcon etc. - I am now adding a manual `run_tests` custom command/target to call `ctest` in my CMakeLists which works around this. I did notice that you used Catch2 for discovery in soss, and that would likely be another solution to this discovery problem.

ipa-mdl, 2021-06-07:
> This is with rosdep.

Did you try https://github.com/ros-industrial/industrial_ci/issues/475#issuecomment-785264919?
ROS1 package discovery is surprisingly complicated..

>  I did notice that you used Catch2 for discovery in soss, and that would likely be another solution to this discovery problem.

Not me, but OSRF.

PR #694 (2021-07-22): Make catkin_tools the default builder for ROS1 noetic
While I would like to use colcon for ROS1 as well, it has several issues.
I hope that this will not break too many builds.
`BUILDER=colcon` is still supported, but now as an opt-in.

closes #629 and closes #666

@wxmerkt: Please test, if this resolves your build issues.
closed 2021-07-22 by ipa-mdl