ros-industrial/industrial_ci #632: cannot run docker image after DOCKER_COMMIT

fmessmer, 2021-02-25:
I ran a travis job locally (`run_travis`) with `DOCKER_COMMIT=debug_mojin`
the travis run succeeded reporting a failure of the source_tests - which was expected and the reason for why I wanted the docker image, i.e. for debugging
the docker image was created, but I cannot run it
```
fxm@alpamayo:~$ docker run -it debug_mojin
/bin/bash: /home/fxm/projects/ci/install/share/industrial_ci/src/run.sh: No such file or directory
```

this has been working for me a few weeks back, but now seems broken
did some of the recent changes break anything?

ipa-mdl, 2021-02-25:
> docker run -it debug_mojin

The created images are not directly runnable (mounts are missing), but meant to be used with `industrial_ci`.
If you just want to inspect the contents, `docker run -it debug_mojin /bin/bash` should give  you shell access.

> the file is present on my laptop, but I don't know why it would need that file on my host to run a container...

#572 changed the mount to have the same path inside Docker as outside (instead of `/root/ici`).
This works better with ISOLATION=shell.

> this has been working for me a few weeks back

I don't see how this could have been working in the past.

ipa-mdl, 2021-02-25:
> docker run -it debug_mojin /bin/bash should give you shell access.

We could implement special support to have an interactive shell including all mounts..

LoyVanBeek, 2021-02-25:
> have an interactive shell including all mounts

That would be quite helpful for debugging. I'm working on this with @fmessmer and just also came to the same conclusion that the `DOCKER_COMMIT` feature does not work in the way we understood (which kinda makes sense in 20/20 hindsight).

ipa-mdl, 2021-02-25:
> came to the same conclusion that the DOCKER_COMMIT feature does not work in the way we understood

The documentation might not cover this aspect sufficiently. (PRs are welcome ^^)

It is meant for use with `rerun_ci` (manages `DOCKER_COMMIT` automatically) or to cache dependencies for subsequent stages/jobs..

fmessmer, 2021-02-25:
ok, then this was probably a usage error on my side
feel free to close this issue or keep as a reminder to update docu (won't be able to contribute here atm)

closed 2021-07-16 by ipa-mdl