ros-industrial/industrial_ci #635: Using industrial_ci with Cloud Build

pedro1713, 2021-03-01:
I'm trying to set up a ROS CI pipeline using Google's Cloud Build, I would like to use industrial_ci but I don't see support for Cloud Build. Is there a plan to support Cloud Build? Or some guidance on how to run the scripts to set up a pipeline under Google's services.

gavanderhoorn, 2021-03-01:
Cross-post of [Setting up industrial_ci inside Google Cloud Build](https://answers.ros.org/question/372900) -- my suggestion.

ipa-mdl, 2021-03-01:
>  I would like to use industrial_ci but I don't see support for Cloud Build.

I don't know Google Cloud build, but if it works with Docker (or isolated) VMs, then it is "supported" by `ci.sh` (only assumption: current directory is target repository).
Additional support (variable mapping, folding) could be added in a specific entry script. 
These entry scripts are only provided for convenience, see minimal [bitbucket.sh](https://github.com/ros-industrial/industrial_ci/blob/master/bitbucket.sh#L25).

pedro1713, 2021-03-01:
When attempting to run ci.sh I get the following errors:
`Step #1: >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Step #1: Starting function 'pull_docker_image'
Step #1: /workspace/.industrial_ci/industrial_ci/src/util.sh: line 164: docker: command not found
Step #1: <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Step #1: Function 'pull_docker_image' returned with code '127' after 0 min 0 sec
Finished Step #1
ERROR
ERROR: build step 1 "ros:melodic" failed: step exited with non-zero status: 127`
Notice that in my cloudbuild.yaml file I run steps with the name 'ros:melodic' which as I understand their documentation means that my steps are run inside a docker image of ROS with melodic tag pulled from DockerHub. The first step in my build actually pulls that image. 
For reference this is currently my configuration file:
`steps:
- name: 'ros:melodic'
  entrypoint: 'bash'
  args:
    - '-c'
    - git clone --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
- name: 'ros:melodic'
  entrypoint: 'bash'
  args:
    - '-c'
    - .industrial_ci/ci.sh`

Any ideas of what could be wrong?

gavanderhoorn, 2021-03-01:
> docker: command not found

is `docker` installed and available to scripts run as part of your build?

pedro1713, 2021-03-01:
I have changed the image I used in the config file from 'ros:melodic' to 'docker:git'. Using that container I have no issues with the docker command. I get the following error now:
Step #1: Already have image: docker:git
Step #1: /usr/local/bin/docker-entrypoint.sh: exec: line 61: git clone --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master: not found
Finished Step #1
ERROR
ERROR: build step 1 "docker:git" failed: step exited with non-zero status: 127

I think my issue is with properly writing multi-line commands in Cloudbuild configuration file.

ipa-mdl, 2021-03-01:
> is docker installed and available to scripts run as part of your build?

Not in `ros:melodic`.

> Any ideas of what could be wrong?

If the outer CI pipeline already pulls the image, you can try `ISOLATION=shell`.

> For reference this is currently my configuration file:

According to https://cloud.google.com/build/docs/configuring-builds/run-bash-scripts,
this should work:

```yaml
steps:
- name: 'ros:melodic'
  entrypoint: 'bash'
  args:
  - '-c'
  - |-
    git clone --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
    .industrial_ci/ci.sh
  env:
  - 'ISOLATION=shell'

```

ipa-mdl, 2021-03-01:
>  I have changed the image I used in the config file from 'ros:melodic' to 'docker:git'

Should work as well, but without `ISOLATION=shell`.

> I think my issue is with properly writing multi-line commands in Cloudbuild configuration file.

Yes, sounds like it..

ipa-mdl, 2021-03-01:
> I think my issue is with properly writing multi-line commands in Cloudbuild configuration file.

```
not found
```

bash is needed to run `industrial_ci` and it is missing from `docker:git`(alpine-based).
Please try adding ` apk add --update bash coreutils tar`.

pedro1713, 2021-03-01:
This is my config file right now:
`steps:
  - name: 'docker:git'
    entrypoint: 'apk'
    args:
      - add
      - --update
      - bash
      - coreutils
      - tar
  - name: 'docker:git'
    entrypoint: 'bash'
    args:
      - '-c'
      - |-
        git clone --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
        .industrial_ci/ci.sh`

I still get an error related to not finding bash in the build. Output from the error below: 
`Step #0: (1/9) Installing readline (8.1.0-r0)
Step #0: (2/9) Installing bash (5.1.0-r0)
Step #0: Executing bash-5.1.0-r0.post-install
Step #0: (3/9) Installing libacl (2.2.53-r0)
Step #0: (4/9) Installing libattr (2.4.48-r0)
Step #0: (5/9) Installing skalibs (2.10.0.0-r0)
Step #0: (6/9) Installing s6-ipcserver (2.10.0.0-r0)
Step #0: (7/9) Installing utmps (0.1.0.0-r0)
Step #0: Executing utmps-0.1.0.0-r0.pre-install
Step #0: (8/9) Installing coreutils (8.32-r2)
Step #0: (9/9) Installing tar (1.33-r1)
Step #0: Executing busybox-1.32.1-r3.trigger
Step #0: OK: 27 MiB in 35 packages
Finished Step #0
Starting Step #1
Step #1: Already have image: docker:git
Finished Step #1
ERROR
ERROR: build step 1 "docker:git" failed: starting step container failed: Error response from daemon: OCI runtime create failed: container_linux.go:370: starting container process caused: exec: "bash": executable file not found in $PATH: unknown`

It shows in step 0 that bash was installed though. And my understanding is that the workspace folder that is setup by Cloud Build should persist between steps. Do I need to do everything in one step? Could that be the case?

ipa-mdl, 2021-03-01:
> And my understanding is that the workspace folder that is setup by Cloud Build should persist between steps. 

The workspace might be persisted, but the environment is not.

> Do I need to do everything in one step? Could that be the case?

Yes, I think so.

pedro1713, 2021-03-01:
My package failed due to a dependency on a proprietary message type for a sensor we used. However, the following configuration script works:
`steps:
  - name: 'ros:melodic'
    entrypoint: 'git'
    args:
      - clone
      - --depth
      - '1'
      - https://github.com/ros-industrial/industrial_ci
      - .industrial_ci
      - -b
      - master
  - name: 'ros:melodic'
    entrypoint: 'bash'
    args:
      - '-c'
      - |-
        git clone --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
        .industrial_ci/ci.sh
    env:
      - 'ISOLATION=shell'
      - 'CI=true'`
I had to move back to ros:melodic image as the lack of bash was giving me trouble in the docker:git image. 
Thanks for all the help.

closed 2021-03-02 by pedro1713

gavanderhoorn, 2021-03-02:
Would be nice to add this to the documentation, next to the description of how to use `industrial_ci` with other providers.

@pedro1713: would you be willing to contribute that in a PR?

pedro1713, 2021-03-02:
Sure, what do I need to do for a proper PR? I can work on that sometime later today.

ipa-mdl, 2021-03-02:
@pedro1713: Great that it works now!  Why do you need two steps?

> Sure, what do I need to do for a proper PR? 

For now, you could add a quickstart example: https://github.com/ros-industrial/industrial_ci/blob/master/README.rst#quick-start

pedro1713, 2021-03-02:
Thanks for pointing that out. I just didn't realize the first step was doing the same as the proposed step, and I left it in the script. The script still works without it. 
I'll look into adding a small section to  https://github.com/ros-industrial/industrial_ci/blob/master/README.rst#quick-start and the corresponding PR