ros-industrial/industrial_ci #639: /home/runner/work/_actions/ros-industrial/industrial_ci/master/.github/action.sh: No such file or directory

cardboardcode, 2021-03-14:
Hi, thank you for developing and maintaining this essential ROS quality assurance tool.

This error is consistently encountered under the following scenario:
:warning: 
```bash
/__w/_temp/c678a811-8165-41e2-a8c1-8a4d2dd6ce5f.sh: line 1: /home/runner/work/_actions/ros-industrial/industrial_ci/master/.github/action.sh: No such file or directory
Error: Process completed with exit code 127.
```

Using **custom public docker** for use in GitHub Actions workflow CI. :whale: 

```yaml
name: CI

on: [push, pull_request]
jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: foxy, ROS_REPO: testing}
          - {ROS_DISTRO: foxy, ROS_REPO: main}
    env:
      CCACHE_DIR: /github/home/.ccache

    runs-on: ubuntu-20.04
    # Use custom public docker image.
    container:
      image: cardboardcode/epd-foxy-base:emd_compatible
    steps:
      - uses: actions/checkout@v2
      - uses: actions/cache@v2
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      # Run industrial_ci
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}

```

Not including the use of the public custom docker image in the above workflow `.yml` file **resolves the error**. 

```yaml
    container:
      image: cardboardcode/epd-foxy-base:emd_compatible
``` 

**However**, the point about using the docker image is to **address external dependencies** needed to build the ROS2 package. Therefore, the custom docker is needed.

ipa-mdl, 2021-03-14:
I am not sure why you get this error.
It might be just the result of some other (hidden) error or bug.
I haven't tested `container/image` yet.

> the point about using the docker image is to address external dependencies needed to build the ROS2 package.

By default `industrial_ci` uses a new Docker container for the test.
You can specify `ISOLATION: shell` to use the host container.

However, in your case the host container does seem to work with your custom image, so you could go with `ISOLATION: docker` (the default) and just specify your custom image with `DOCKER_IMAGE: cardboardcode/epd-foxy-base:emd_compatible`.

ipa-mdl, 2021-03-14:
I ran some tests and I can confirm the issue.
It looks like there is a bug in GitHub actions with custom docker images and composite actions.
In this case the path gets altered (`-v "/home/runner/work/_actions":"/__w/_actions"`), and does not match [github.action_path](https://github.com/ros-industrial/industrial_ci/blob/2867438f4964cd217c29d219e53f7f93d53602ca/action.yml#L9) anymore.

ipa-mdl, 2021-03-14:
I will try this: https://github.community/t/failing-composite-action-verbatim-from-docs/160342

ipa-mdl, 2021-03-14:
With #640 you will get past the "No such file or directory" error (https://github.com/ipa-mdl/ci_tests/runs/2107060651), but it will stop with

>   /__w/_actions/ipa-mdl/industrial_ci/action-debug/industrial_ci/src/util.sh: line 164: docker: command not found

This is expected, because is not part of your custom image.
Just use `ISOLATION: shell` or remove `container/image` and set `DOCKER_IMAGE` instead.

ipa-mdl, 2021-03-14:
One last comment: If you use your own (ROS-based) docker image, the `ROS_REPO` will be ignored.

cardboardcode, 2021-03-14:
Thank you. @ipa-mdl 

Will test it out once my bandwidth allows. Thank you so much for this in-depth and prompt investigation into the issue.

closed 2021-03-14 by ipa-mdl

cardboardcode, 2021-03-14:
Verified that the above **pull request** and **suggested edit** helped resolved the issue in this thread.

## Solution
For anyone who is facing this issue in the future, just for reference, my workflow `.yml` file now looks like this:

```yaml
name: CI

on: [push, pull_request]
jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: foxy, ROS_REPO: testing}
          - {ROS_DISTRO: foxy, ROS_REPO: main}
    env:
      CCACHE_DIR: /github/home/.ccache
      ISOLATION: shell

    runs-on: ubuntu-20.04
    # Use custom public docker image.
    container:
      image: cardboardcode/epd-foxy-base:emd_compatible
    steps:
      - uses: actions/checkout@v2
      - uses: actions/cache@v2
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      # Run industrial_ci
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}
```

PR #640 (2021-03-14): Use GITHUB_ACTION_PATH instead of github.action_path
Fixes #639 and helps with #638