ros-industrial/industrial_ci #641: CMake error when enabling ccache on GitHub Actions

tylerjw, 2021-03-18:
I get this cmake error when enabling ccache on github actions:

```
Starting >>> random_numbers
  --- stderr: random_numbers
  CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
    The C compiler
  
      "/usr/lib/ccache/cc"
  
    is not able to compile a simple test program.
```

Has anyone else seen this?

tylerjw, 2021-03-20:
I was using the cache directory step wrong.  I tried using the $HOME environment variable.  The current example has a full path to a directory and the github documentation discourages that.  I've switched to using `~/.ccache` and that seems to work.  However I'm now trying to debug if it is actually working because when I restore a cache it always seems to be 22 bytes which makes me think it is actually not working.

Does anyone have a current GitHub Actions configuration using ccache I could try to copy from?

tylerjw, 2021-03-20:
For context here is my current yml I'm using:

```
# This config uses industrial_ci (https://github.com/ros-industrial/industrial_ci.git).
# For troubleshooting, see readme (https://github.com/ros-industrial/industrial_ci/blob/master/README.rst)

name: BuildAndTest

on: [push, pull_request]

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: melodic, ROS_REPO: main, CATKIN_LINT: pedantic}
    env:
      CCACHE_DIR: ~/.ccache
      UPSTREAM_WORKSPACE: https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall

    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
      - uses: actions/cache@v2
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}-${{ github.sha }}  
          restore-keys: |
            ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}
```

Also note that I'm using the restore-keys feature because my understanding is that if you get an exact keys match it won't upload at the end.  In that case it will only download the cache with that key before the job.  This seems like it would make ccache less and less useful as time goes on.  That being said, I've never seen it restore when it didn't get an exact match.

tylerjw, 2021-03-20:
Here is an example of the ccache dir line in the example not working: https://github.com/ros-planning/warehouse_ros/pull/64/checks?check_run_id=2156480871

When you use this: `CCACHE_DIR: /github/home/.ccache`

you get a bunch of errors that look like this:

```
Run actions/cache@v2
  with:
    path: /github/home/.ccache
    key: ccache-foxy-main-2507f8ca0f0122cd9cf7148e169f9df5336d234c
    restore-keys: ccache-foxy-main
  
  env:
    CCACHE_DIR: /github/home/.ccache
Received 532048 of 532048 (100.0%), 2.5 MBs/sec
Cache Size: ~1 MB (532048 B)
/usr/bin/tar --use-compress-program zstd -d -xf /home/runner/work/_temp/485660a6-4093-4d9f-8eea-c44fbdc28649/cache.tzst -P -C /home/runner/work/warehouse_ros/warehouse_ros
/usr/bin/tar: ../../../../../github: Cannot mkdir: Permission denied
/usr/bin/tar: ../../../../../github/home/.ccache: Cannot mkdir: No such file or directory
/usr/bin/tar: ../../../../../github: Cannot mkdir: Permission denied
/usr/bin/tar: ../../../../../github/home/.ccache/5: Cannot mkdir: No such file or directory
/usr/bin/tar: ../../../../../github: Cannot mkdir: Permission denied
/usr/bin/tar: ../../../../../github/home/.ccache/5/stats: Cannot open: No such file or directory
/usr/bin/tar: ../../../../../github: Cannot mkdir: Permission denied
/usr/bin/tar: ../../../../../github/home/.ccache/5/5: Cannot mkdir: No such file or directory
/usr/bin/tar: ../../../../../github: Cannot mkdir: Permission denied
/usr/bin/tar: ../../../../../github/home/.ccache/5/5/d176845cb06639f4aea2bfb51893e0-1539.manifest: Cannot open: No such file or directory
...
```

tylerjw, 2021-03-20:
I think this might be a bug in `industrial_ci` because when I specify it this way:

```
jobs:
  industrial_ci:
    strategy:
      matrix:
        ros_distro: [ foxy ]
        ros_repo: [ main, testing ]
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
      - name: install ccache
        run: sudo apt -qq install ccache
      - uses: actions/cache@v2
        with:
          path: ~/.ccache
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}-${{ github.sha }}  
          restore-keys: |
            ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      - name: ccache stats
        run: ccache -s
      - uses: 'ros-industrial/industrial_ci@master'
        env: 
          ROS_DISTRO: ${{ matrix.ros_distro }}
          ROS_REPO: ${{ matrix.ros_repo }}
          CCACHE_DIR: ~/.ccache
          AFTER_BUILD_TARGET_WORKSPACE: 'ccache -s'
      - name: ccache stats
        run: ccache -s
```

The output inside the industrial_ci run looks like this (after build target workspace):

```
   cache directory                     /home/runner/work/warehouse_ros/warehouse_ros/~/.ccache
  primary config                      /home/runner/work/warehouse_ros/warehouse_ros/~/.ccache/ccache.conf
  secondary config      (readonly)    /etc/ccache.conf
  stats updated                       Sat Mar 20 19:40:37 2021
  cache hit (direct)                     1
  cache hit (preprocessed)               0
  cache miss                            10
  cache hit rate                      9.09 %
  called for link                       12
  cleanups performed                     0
  files in cache                        22
  cache size                           3.6 MB
  max cache size                       5.0 GB
```

And then the one out in the runner later looks like this:
```
 cache directory                     /home/runner/.ccache
primary config                      /home/runner/.ccache/ccache.conf
secondary config      (readonly)    /etc/ccache.conf
cache hit (direct)                     0
cache hit (preprocessed)               0
cache miss                             0
cache hit rate                      0.00 %
cleanups performed                     0
files in cache                         0
cache size                           0.0 kB
max cache size                       5.0 GB
```

Here is the job: https://github.com/ros-planning/warehouse_ros/runs/2156701400
The PR where I've tried a bunch of combinations to figure out what works: https://github.com/ros-planning/warehouse_ros/pull/64

In short I think this configuration works:
```
jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: foxy, ROS_REPO: main}
          - {ROS_DISTRO: foxy, ROS_REPO: testing}
    env:
      CCACHE_DIR: /home/runner/.ccache
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
      - uses: actions/cache@v2
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}-${{ github.sha }}  
          restore-keys: |
            ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}
```

ipa-mdl, 2021-03-20:
>           CCACHE_DIR: ~/.ccache

This will not work, because there is no way to expand `~` in a string like this.
And I think the cache directory must be inside the build directory (at least in the past).
Did you try `CCACHE_DIR=.ccache`?

tylerjw, 2021-03-20:
> Did you try `CCACHE_DIR=.ccache`?

I did, this causes the same permissions errors you see in my first post here where ccache errors because the workspace directory where the repo is checked out is a read-only filesystem.

tylerjw, 2021-03-20:
Also note that `$HOME/.ccache`, `${HOME}/.ccache`, '${{ HOME }}/.ccache`,  '/$HOME/.ccache` are all other combinations I tried.  I also tried various quotes around them to try to see if the expansion of the $HOME variable could be affected.  In all these cases it does seem like it is extracting the cache to the correct location (in the home directory) on the host runner.  However when it is mounting into the docker image industrial_ci is treating these paths as if they are relative paths under the repo.

tylerjw, 2021-03-20:
On the other hand if I specify the absolute path of `/home/runner/.ccache` everything works.  I think we should just update the example for GitHub actions to encourage that path instead of `/github/home/.ccache` which is no longer correct.

gavanderhoorn, 2021-03-21:
I've been using this (from [here](https://github.com/ros-industrial/fanuc/blob/3ea2842baca3184cc621071b785cbf0c588a4046/.github/workflows/ci_bionic.yml#L23)):

```yaml
    env:
      CCACHE_DIR: "${{ github.workspace }}/.ccache"
```

afaict, that works. It's sort-of an absolute path, but not quite.

ipa-mdl, 2021-03-22:
`CCACHE_DIR: "${{ github.workspace }}/.ccache"` is better than anything in the home directory.
I thought that `CCACHE_DIR: .ccache` would resolve to  this directory automatically, I will double check.

tylerjw, 2021-03-22:
Why is that better than the home directory?

Is it because the home directory could change?

ipa-mdl, 2021-03-22:
> Why is that better than the home directory?

IMHO the ccache is workspace-specific data and not user-specific, especially with self-hosted runners in mind.
In addition the home directory would tempt the user to refer to it with `~`, which would need special support..

closed 2021-03-25 by tylerjw