ros-industrial/industrial_ci #642: ros2 run rerun_ci not working (missing file in container)

tylerjw, 2021-03-18:
There is some file not being copied into the container is my suspicion for this.  I'm testing converting `warehouse_ros_mongo` to industrial_ci on GitHub Actions.  [See my PR here](https://github.com/ros-planning/warehouse_ros_mongo/pull/43)

I'm having some issue on the foxy/main that doesn't fail on foxy/testing so I'd like to be able to run locally.  I used colcon to build industrial_ci and then sourced my local workspace and did this:

```bash
$ ros2 run industrial_ci rerun_ci src/warehouse_ros_mongo ROS_DISTRO=foxy ROS_REPO=main CCACHE_DIR="$HOME/.ccache"
Testing /home/tyler/code/moveit2/ws_moveit2/src/warehouse_ros_mongo
Running test 'source_tests'
Copy credentials: /home/tyler/.docker
Copy credentials: /home/tyler/.ssh
/bin/bash: /home/tyler/code/moveit2/ws_moveit2/install/industrial_ci/share/industrial_ci/src/run.sh: No such file or directory
Committing container to tag: 'industrial-ci/rerun_ci/warehouse_ros_mongo:f3cdd18ec90d'
Tests failed exit code '127'
Please do not forget to clean-up: docker rmi industrial-ci/rerun_ci/warehouse_ros_mongo:f3cdd18ec90d
```

That file does exist on my local filesystem, I image the issue is actually about it not existing on the docker image.  Or am I just using it wrong?

tylerjw, 2021-03-18:
I got it working with catkin and rosrun from noetic.  Looking at the documentation there is no indication that this will work with ros2 run.  In the future it might be worth supporting this as more people are using ros2.

ipa-mdl, 2021-03-19:
Thanks for reporting!
In theory it should work the same way for `rosrun` and  `ros2 run`.

> That file does exist on my local filesystem, I image the issue is actually about it not existing on the docker image. 

It should exist, because it gets bind-mounted.

ipa-mdl, 2021-03-19:
@tylerjw: How *exactly* did you build the workspace?

tylerjw, 2021-03-19:
Created this file structure:

```
ws_testing/src/industrial_ci - clone of master
ws_testing/src/warehouse_ros_mongo - what I want to test
```

In the `ws_testing` folder I ran these commands:
```
source /opt/ros/foxy/setup.zsh
colcon build --symlink-install --packages-select industrial_ci
source install/setup.zsh
```

Then I tried to run the test like this:
```
ros2 run industrial_ci rerun_ci src/warehouse_ros_mongo ROS_DISTRO=foxy ROS_REPO=main
```

This was the output:
```
Testing /home/tyler/code/moveit2/ws_testing/src/warehouse_ros_mongo
Running test 'source_tests'

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'pull_docker_image'
focal: Pulling from library/ubuntu
Digest: sha256:b4f9e18267eb98998f6130342baacaeb9553f136142d40959a1b46d6401f0f2b
Status: Image is up to date for ubuntu:focal
docker.io/library/ubuntu:focal
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'pull_docker_image' returned with code '0' after 0 min 0 sec
Copy credentials: /home/tyler/.docker
Copy credentials: /home/tyler/.ssh
/bin/bash: /home/tyler/code/moveit2/ws_testing/install/industrial_ci/share/industrial_ci/src/run.sh: No such file or directory
Committing container to tag: 'industrial-ci/rerun_ci/warehouse_ros_mongo:890bd59f16c3'
Tests failed exit code '127'
Please do not forget to clean-up: docker rmi industrial-ci/rerun_ci/warehouse_ros_mongo:890bd59f16c3
```

tylerjw, 2021-03-19:
Maybe this is caused by `--symlink-install`.

```
ll /home/tyler/code/moveit2/ws_testing/install/industrial_ci/share/industrial_ci/src/run.sh
lrwxrwxrwx 1 tyler tyler 78 Mar 19 13:00 /home/tyler/code/moveit2/ws_testing/install/industrial_ci/share/industrial_ci/src/run.sh -> /home/tyler/code/moveit2/ws_testing/src/industrial_ci/industrial_ci/src/run.sh
```

tylerjw, 2021-03-19:
I just re-ran it without `--symlink-install` and it works.  Maybe just a note in the docs would be good.  I'm not sure if --symlink-install could be made to work.  I'm sorry I didn't realize earlier that was what caused it.

ipa-mdl, 2021-03-19:
> Maybe this is caused by --symlink-install.

This is what I suspected.
However, it works with ROS2 eloquent..
I guess something has changed with Ubuntu focal (or ROS2 foxy).

> I'm not sure if --symlink-install could be made to work

It should be possible..

tylerjw, 2021-03-19:
I just tested on `rolling` and it doesn't work there either.  I suspect something changed in foxy.

PR #643 (2021-03-19): Resolve all symbolic links before starting ci_main.sh from run_ci
closes #642 

The issue with `--symlink-install` is that the files are linked individually, but the installed `src` folder actually exists.
This fix changes to real `src` first.

@tylerjw: please test
closed 2021-03-19 by ipa-mdl