ros-industrial/industrial_ci #644: Uploading test artifacts to GitHub Actions

tylerjw, 2021-03-20:
GitHub Actions has a feature for uploading test artifacts: https://docs.github.com/en/actions/guides/storing-workflow-data-as-artifacts

It would be really neat if we could use this to upload the xml result files from tests.  To do this we'd need to copy those files out of the docker image so the upload artifacts github action could get to them.  I'm not sure where to start to try to do this.

ipa-mdl, 2021-03-20:
You could just set `BASEDIR` (e.g. `.base`) and will have all workspace s(e.g,  `.base/target_ws` in there.
Docker will be set-up automatically to bind-mount this directory.

tylerjw, 2021-03-20:
Thank you for that.  I got it working like this (uploading artifacts on test failure):

```
name: BuildAndTest

on: [push, pull_request]

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: foxy, ROS_REPO: main}
          - {ROS_DISTRO: foxy, ROS_REPO: testing}
    env:
      CCACHE_DIR: /home/runner/.ccache
      BASEDIR: .base
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
      - uses: actions/cache@v2
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}-${{ github.sha }}  
          restore-keys: |
            ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}
      - uses: actions/upload-artifact@v2
        if: failure()
        with:
          name: test-results
          path: ${{ env.BASEDIR }}/target_ws/**/test_results/**/*.xml
```

Should I make a PR to add this to the "full" github actions example?

rhaschke, 2021-04-10:
@tylerjw, I think this is solved and could be closed?

closed 2021-04-10 by tylerjw