ros-industrial/industrial_ci #647: CCACHE_SIZE enviroment variable

tylerjw, 2021-03-22:
It would be useful to be able to set the CCACHE_SIZE environment variable for reducing the size of the cache sometimes.

ipa-mdl, 2021-03-22:
At first I thought that this is a variable that ccache support directly, but I guess it has to be [set explicitly](https://wiki.archlinux.org/index.php/ccache#Set_maximum_cache_size).

It would be great, if you could create a PR and test it.

Instead of adding it everywhere, I would suggest a new function in workspace.sh:

```bash
function ici_setup_ccache() {
    if [ -n "$CCACHE_DIR" ]; then
        ici_run "setup_ccache" ici_apt_install ccache
        export PATH="/usr/lib/ccache:$PATH"
        if [ -n "$CCACHE_SIZE" ]; then
            ccache --set-config=max_size="$CCACHE_SIZE"
        fi
    fi
}
```
(not tested)

`CCACHE_SIZE` needs to be added in docker.env as well...

tylerjw, 2021-03-22:
Basically the reason for wanting this feature is debug builds can create a huge ccache that can fill up storage and you get still get a large benefit by setting it to something small-ish (but still big enough for anything you are going to get a hit on) (like 500mb) so it cleans out the old stuff sooner.