ros-industrial/industrial_ci #650: Function 'install_upstream_dependencies' returned with code '1' after 0 min 0 sec

130s, 2021-03-27:
`install_upstream_dependencies` job keeps failing instantly.

Ex. on [my GitHub Actions job](https://github.com/130s/docker_vcstool/pull/2/checks?check_run_id=2208111145#step:3:808)

```
  Starting function 'install_upstream_dependencies'
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'install_upstream_dependencies' returned with code '1' after 0 min 0 sec
Error: Process completed with exit code 1.
```

<details><summary>Btw looking into the ICI's code (feeling like I'm back home :), I can't find where that function is defined.</summary>

```
$ git checkout master && git log -1 --oneline 
Already on 'master'
Your branch is up to date with 'origin/master'.
bfc68c8 (HEAD -> master, origin/master) Resolve all symbolic links before starting ci_main.sh from run_ci

$ ack -ir install_upstream .
$ 
```
</details>

ipa-mdl, 2021-03-27:
> Ex. on my GitHub Actions job for this PR 130s/docker_vcstool#2:

These are private (?)

>  I can't find where that function is defined.

It gets prefix by the workspace, which gets build, see https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/workspace.sh#L324.

jdlangs, 2021-07-16:
To add a datapoint here, I got this error with no further log output. It happened I had a malformed `package.xml` in one of my upstream repos which I believe breaks the dependency tooling. I have no idea if that's universally the problem but it seems to be at least one potential root issue.

ipa-mdl, 2021-07-16:
> It happened I had a malformed package.xml in one of my upstream repos which I believe breaks the dependency tooling. 

Good point! I face something similar yesterday as well.

> I have no idea if that's universally the problem but it seems to be at least one potential root issue.

The rosdep output gets filtered heavily (with grep), I will try to find a different solution.

ipa-mdl, 2021-07-16:
https://github.com/ros-industrial/industrial_ci/pull/709 should show the rosdep errors
@130s, @jdlangs: Please test this case again with the latest version after it was merged.

130s, 2021-07-16:
I'd love to but it's unclear if I get to in reasonably soon.

jdlangs, 2021-07-21:
I can confirm in the case of the bad `package.xml` I now see a useful error message in the log output.

ipa-mdl, 2021-07-21:
@jdlangs: Thanks!

I will close this issue now.

closed 2021-07-21 by ipa-mdl