ros-industrial/industrial_ci #656: Output with many empty lines

rhaschke, 2021-04-10:
With `VERBOSE_OUTPUT=true` and `BUILDER=catkin_tools` I observe [very sparse output (every 2nd line is empty)](https://github.com/rhaschke/rviz/runs/2314728940?check_suite_focus=true):
```
  make[2]: Entering directory '/home/runner/work/target_ws/build/rviz'
  
  [  0%] Built target tf_generate_messages_lisp
  
  [  1%] Automatic MOC for target connect_test
  
  /usr/bin/make -f src/test/CMakeFiles/interactive_marker_test_autogen.dir/build.make src/test/CMakeFiles/interactive_marker_test_autogen.dir/depend
  
  cd /home/runner/work/target_ws/build/rviz/src/test && /usr/bin/cmake -E cmake_autogen /home/runner/work/target_ws/build/rviz/src/test/CMakeFiles/connect_test_autogen.dir/AutogenInfo.json ""
```
I'm not used to see such behaviour in the console. Any idea where this comes from?
By the way, why are using immediate, but interleaved output (`-i`) in this mode:
https://github.com/ros-industrial/industrial_ci/blob/d258da3c221a7931b9cdf807f499ad50d847372e/industrial_ci/src/builders/catkin_tools.sh#L38

gavanderhoorn, 2021-04-11:
Isn't this https://github.com/catkin/catkin_tools/issues/643?

rhaschke, 2021-04-11:
You are right. Can be closed as duplicate of https://github.com/catkin/catkin_tools/issues/643.
My other question remains: Why using `-vi`?

gavanderhoorn, 2021-04-11:
> My other question remains: Why using `-vi`?

I have a vague recollection this has to do with visibility of desired bits of output, but @ipa-mdl would have the authoritative answer here.

I'd also suggest to update the issue title, as it's no(t) (longer) about empty lines in the log any more.

ipa-mdl, 2021-04-19:
> My other question remains: Why using -vi?

Not 100% sure, it might have historic reasons ;)
One advantage of `-vi` is the immediate response, e.g. with `run_ci`.
Or if the build/tests take very long and the overall CI timeout is reached.

ipa-mdl, 2021-04-19:
> it might have historic reasons

It has been introduced like this in #1 :D
And #94 added the option `VERBOSE_OUTPUT` to turn it off.

rhaschke, 2021-04-19:
Ok. Maybe, interleaved output was never an issue because the CI runners only provide a single processor?

closed 2021-04-19 by rhaschke