ros-industrial/industrial_ci #657: ici environment variables accumulate within docker image

rhaschke, 2021-04-11:
Using `DOCKER_COMMIT` and some tricks, I managed to cache the docker image for my `ici` builds. This is particularly useful, because  `install_<workspace>_dependencies` often takes more time than the (ccached) build.
However, I'm facing a problem when switching configurations: Environment variables that became undefined are still present in the docker image, because `ici` doesn't clear the variables listed in [`docker.env`](https://github.com/ros-industrial/industrial_ci/tree/master/industrial_ci/src/isolation/docker.env).

It would be also great to allow a more fine-grained access to individual build steps for caching. Did you already had a look at [`docker buildx`](https://github.com/docker/buildx)? They seem to have the possibility of caching images also _locally_ instead of using dockerhub. This could be used to save those images via a CI provider's cache. [Here is an example for GHA](https://github.com/docker/build-push-action/blob/master/docs/advanced/cache.md#github-cache).
If docker `ISOLATION` would create/build a sequence of docker images for all build steps, we could individually decide, which of those to make accessible or cache. [Here is a (very first) attempt into that direction.](https://github.com/ros-industrial/industrial_ci/compare/master...rhaschke:docker-buildx)

rhaschke, 2021-04-11:
Having thought more about it, the main reason to get more fine-grained access to build steps is not for caching, but for accessing the results of individual steps, on which basis we can trigger then various other steps within e.g. uploading test results (#644).
Thus, we don't need to split up the build process into smaller steps, but just make the results available into an environment file (`$GITHUB_ENV` for GHA).

Caching of the isolating docker container should only comprise the installed prerequisites. My current caching approach [pulls and pushes the committed docker on _each_ CI run](https://github.com/ros-visualization/rviz/blob/melodic-devel/.github/workflows/ci.yaml#L58-L72), which obviously accumulates layers. 
As far as I understand, defining a `BASEDIR` and `CCACHE_DIR` bind-mounts all folders producing build artifacts from the host system, so their content doesn't populate image layers. Nevertheless, the number of layers continuously increases over time, which obviously should be avoided. Also, we need a mechanism to monitor the list of prerequisite packages (and their versions?) installed by an `ici` run. This could be handled by a file again:
- ICI writes a list of all installed pkgs (can be easily done as part of `ici_install_pkgs_for_command`)
- We cache this file between CI runs (using a key that encodes all CI config options that might affect this list)
- If, after a CI run, the cached file and the new file have differences, the cached docker image should be invalidated, i.e. removed from dockerhub and a new image should be build (and pushed) based on the logged list of packages
- Maybe we should log a list of shell commands directly, e.g. to setup ROS repos etc.

rhaschke, 2021-04-11:
@tylerjw: this thread will be interesting for you :wink:

ipa-mdl, 2021-04-15:
> It would be also great to allow a more fine-grained access to individual build steps for caching

So in way, you want to implement [ros_buildfarm](https://github.com/ros-infrastructure/ros_buildfarm)? ;)

> We would need a mechanism to monitor the list of prerequisite packages installed by an ici run and push a new docker commit only if there are changes. 

I was thinking of something similar..
The idea would be to implement a `bundle_dependencies` script:
I would use `workspace.sh` in temp folder/container, download the code and resolve the rosdep keys, which will be feed into a Dockerfile, which will install these dependencies and spit out an image with just the dependencies and nothing else (env, src, etc.). We just have to find a way to break the cache, if needed.

This image will then  be pushed and used in subsequent industrial run.

`docker buildx` is very useful, but I would rely on it yet, because it might not be deployed to all CI services and runners.
In the past I had caching issues with buildkit and multistage Dockerfiles, but I hope that these have been fixed in the recent version.

> because ici doesn't clear the variables listed in docker.env.

This should not be much of issue, because subsequent run will overwrite then variables again.

rhaschke, 2021-04-15:
No, I don't want to reimplement the ROS buildfarm and actually the buildfarm is rather poor at caching...

> We just have to find a way to break the cache, if needed.

This is trivial on GHA: Just use [`${{ hashfiles(workspace.sh) }}`](https://docs.github.com/en/actions/reference/context-and-expression-syntax-for-github-actions#hashfiles). I'm sure other CI providers have similar mechanisms.

> > because ici doesn't clear the variables listed in docker.env.
> 
> This should not be much of issue, because subsequent run will overwrite then variables again.

Sure, not if we have the proposed mechanism in place. Currently, it _is_ a problem.

ipa-mdl, 2021-04-26:
> This is trivial on GHA: Just use ${{ hashfiles(workspace.sh) }}. I'm sure other CI providers have similar mechanisms.

I did not mean to break the cache because of changes in `industrial_ci`, but any upstream change.
Gitlab CI has a button to clear all caches.

ipa-mdl, 2021-04-26:
I have just created a proof-of-concept script: https://gist.github.com/ipa-mdl/575f9b84ed039ff86b74f14c317de2cf
It builds a noetic-based Docker image (named "baked")  with all rosdeps of the input directoy installed.

Under the hood the script calls `buildx` with a generated multi-stage Dockerfile.
The first stages clones industrial_ci, prepares `TARGET_WORKSPACE` and collects the apt package names from rosdep into `/root/src/packages`. The second stage just consumes this file by installing all packages.

This should better be implemented in Python, as a separate Github action.

rhaschke, 2021-04-26:
Thanks for your initiative! I need to have a look later this week...

> > This is trivial on GHA: Just use `${{ hashfiles(workspace.sh) }}`. I'm sure other CI providers have similar mechanisms.
> 
> I did not mean to break the cache because of changes in `industrial_ci`, but any upstream change.

With `workspace.sh` I wanted to refer to the generated script to install all upstream dependencies. I didn't mean to refer to any file in `industrial_ci`. I just noticed that this was confusing because industrial_ci has a `workspace.sh` file. Sorry.