ros-industrial/industrial_ci #661: Large copy when BASEDIR inside repo dir on GitLab

tylerjw, 2021-04-13:
As a result of these lines:

https://github.com/ros-industrial/industrial_ci/blob/bfc68c8a3d5adf12ea4074c2149582398b5b6b54/industrial_ci/src/ci_main.sh#L44-L48

This `.gitlab-ci.yml` results in a huge copy when trying this:

```
stages:
  - test

test:
  except:
    - schedules
  stage: test
  variables:
    ROS_DISTRO: melodic
    ISOLATION: shell
    # repos file containing only exact hash of repos (this almost never changes)
    UPSTREAM_WORKSPACE: 'upstream-fixed.repos'
    BASEDIR: ${CI_PROJECT_DIR}/.workspaces
    CCACHE_DIR: ${CI_PROJECT_DIR}/.ccache
  cache:
    when: always
    key:
      files:
        - upstream-fixed.repos
    paths:
      - .ccache/
      - .workspaces/
  script:
    - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
    - .industrial_ci/gitlab.sh
```

This seemed clever until I realized that when the BASEDIR is defined, the target repo directory is copied into it's own directory.  This results in a huge (and time consuming) copy in this use case.  Maybe the copy could somehow skip the BASEDIR if it is defined within the target repo directory.

This is because of the (imo lame and broken) policy of GitLab only caching directories inside the repo directory.

The workaround I tried was setting the base_dir out of the repo, but still caching in the repo and moving it out before starting industrial ci.  The issue with that is that it updates the timestamps so the build upstream workspace step still happens as if the cache wasn't there.

Do you know a workaround for this or is there someway I could update this step to make it not copy CCACHE or BASEDIR if it is within the target repo.

ipa-mdl, 2021-04-15:
The issue is that the target repo code has to be copied into the target source space (withing BASEDIR).
The latter done to support writing to the source folder (without mutating the code outside, e.g. with `run_ci`).

Perhaps we can exclude BASEDIR and CCACHE from that copy.

ipa-mdl, 2021-04-15:
> Perhaps we can exclude BASEDIR and CCACHE from that copy.

Somehing like:
```
find -maxdepth 1 -mindepth 1 -not -path .git -exec cp -a {} /tmp/random/my_repo \;
```

tylerjw, 2021-04-15:
That would probably work and fix this issue.  I created this workaround for now but I can make a PR with that line to test it too.

```
test:
  stage: test
  variables:
    ROS_DISTRO: melodic
    ISOLATION: shell
    UPSTREAM_WORKSPACE: 'upstream-fixed.repos'
    CACHE_DIR: ${CI_PROJECT_DIR}/.cache
    CCACHE_DIR: ${CI_BUILDS_DIR}/ccache
    BASEDIR: ${CI_BUILDS_DIR}/workspaces
  cache:
    when: always
    key:
      files:
        - upstream-fixed.repos
    paths:
      - .cache
  before_script:
    - mkdir -p ${CCACHE_DIR}
    - mkdir -p ${BASEDIR}
    # Move the cached directories out of the project directory
    - >
      if [ -d ${CACHE_DIR} ]; then
        echo "Moving cached directories out of CI_PROJECT_DIR"
        cp -r -p ${CACHE_DIR}/ccache ${CCACHE_DIR}
        cp -r -p ${CACHE_DIR}/upstream_ws ${BASEDIR}/upstream_ws
        rm -rf ${CACHE_DIR}
      fi
    # Clone industrial_ci
    - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
  script:
    - .industrial_ci/gitlab.sh
  after_script:
    # Copy these directories back into the project directory for caching
    - mkdir -p ${CACHE_DIR}
    - cp -r -p ${CCACHE_DIR} ${CACHE_DIR}/ccache
    - cp -r -p ${BASEDIR}/upstream_ws ${CACHE_DIR}/upstream_ws
    # Copy the test results for artifacts
    - mkdir -p ${CI_PROJECT_DIR}/test_results
    - cp ${BASEDIR}/target_ws/build/*/test_results/*/*.xml ${CI_PROJECT_DIR}/test_results/
    - cp -r ${BASEDIR}/target_ws/logs ${CI_PROJECT_DIR}/test_results/logs
  artifacts:
    when: always
    paths:
      - ${CI_PROJECT_DIR}/test_results/*
    reports:
      junit: ${CI_PROJECT_DIR}/test_results/*.xml
```

Downside to my workaround is that moving these large directories around takes a bunch of time (adds 10min to the ci run) but it doesn't run out of disk space.

tylerjw, 2021-04-16:
> Somehing like:
> 
> ```
> find "$TARGET_REPO_PATH" -maxdepth 1 -mindepth 1 -not -path "$BASEDIR" -exec cp -a {} "$tmp/$(basename "$TARGET_REPO_PATH
> ```

Unfortunately that has the problem of if either the ccache directory or the base path directory is one level deep they get coppied into the base of the repo (out of the sub directory they were in).  I find `find` syntax really confusing and was not able to adapt this to work the way I wanted.  Instead I found a solution of using tar to create a temporary (in memory) archive and extract it.  This doesn't use compression and is nearly as fast as cp.  I've done that in my PR.  Here is a snippet of what I've done:

```bash
function ici_relocate_target_path {
  local tmp_src
  ici_make_temp_dir tmp_src
  mkdir -p "$tmp_src/$(basename "$TARGET_REPO_PATH")"
  [ -n "$BASEDIR" ] && echo Signature: 8a477f597d28d172789f06886806bc55 > "${BASEDIR}/CACHEDIR.TAG"
  [ -n "$CCACHE_DIR" ] && echo Signature: 8a477f597d28d172789f06886806bc55 > "${CCACHE_DIR}/CACHEDIR.TAG"
  tar cf - --exclude-caches-all -C "$TARGET_REPO_PATH" . | tar xf - -C "$tmp_src/$(basename "$TARGET_REPO_PATH")"
  export TARGET_REPO_PATH
  TARGET_REPO_PATH="$tmp_src/$(basename "$TARGET_REPO_PATH")"
}
```

I saw a bunch of examples online of using `rsync` to do this, however rsync would then need to be installed in many of the environments.  `tar` is installed on every distribution we support by default.

ipa-mdl, 2021-04-18:
> tar is installed on every distribution we support by default.

That's not entirely true.  This code is run in the host environment and for Gitlab it depends on the configured image.
`docker:git` (from our default config) uses Alpine, which ships a different tar version. However, our default config installs GNU tar.

PR #664 (2021-04-22): Skip BASEDIR, CCACHE_DIR and industrial_ci in ici_import_directory
reverts #595 and #626

fixes #661

Gitlab: https://gitlab.com/ipa-mdl/industrial_ci/-/pipelines/290145549
closed 2021-04-22 by ipa-mdl