ros-industrial/industrial_ci #666: colcon test-result missing in pre-release job on Noetic

tylerjw, 2021-04-30:
I get this log at the bottom of a pre-release job on noetic for moveit:

```
2021-04-30T17:05:46.0978485Z Summary: 29 packages finished [21min 44s]
2021-04-30T17:05:46.0979913Z   5 packages had stderr output: moveit_commander moveit_ros_move_group moveit_ros_planning moveit_ros_planning_interface pilz_industrial_motion_planner
2021-04-30T17:05:46.0981585Z   5 packages had test failures: moveit_commander moveit_ros_move_group moveit_ros_planning moveit_ros_planning_interface pilz_industrial_motion_planner
2021-04-30T17:05:46.1465236Z # END SUBSECTION
2021-04-30T17:05:46.3839544Z /tmp/tmp.LDeA5yYfRC
2021-04-30T17:05:46.3840294Z # END SECTION
2021-04-30T17:05:46.3840598Z 
2021-04-30T17:05:46.3840973Z Build step 10
2021-04-30T17:05:46.3841269Z 
2021-04-30T17:05:46.3841652Z Build step 11
2021-04-30T17:05:46.3842245Z # BEGIN SECTION: Create collated test stats dir
2021-04-30T17:05:46.3900397Z # END SECTION
2021-04-30T17:05:46.3901118Z 
2021-04-30T17:05:46.3901489Z 
2021-04-30T17:05:46.3902409Z Test results of underlay workspace
2021-04-30T17:05:46.3903097Z 
2021-04-30T17:05:46.7367815Z 'colcon test-result' not available. Please make sure the necessary extension is installed to output the test result summary.
2021-04-30T17:05:46.7368722Z 
2021-04-30T17:05:46.7368965Z 
2021-04-30T17:05:46.7387241Z Skipping empty overlay workspace
```

The pre-release docker image should probably have colcon installed in it?

ipa-mdl, 2021-05-01:
Thanks for reporting!
Could you give me a link to a full log?
According to the code `colcon` should get installed, if necessary..

ipa-mdl, 2021-05-01:
This seems to be an issue with `colcon`.
`colcon test-result --help` actually fails with
```
usage: colcon test-result [-h] [--test-result-base TEST_RESULT_BASE] [--all]
                          [--result-files-only] [--verbose] [--delete]
                          [--delete-yes]
colcon test-result: error: argument --test-result-base: Path 'build' does not exist
```

ipa-mdl, 2021-05-01:
> This seems to be an issue with colcon.

https://github.com/colcon/colcon-test-result/issues/36

tylerjw, 2021-05-01:
Here is a job where it happens.  Because this is a pre-release job of the moveit library the best way to view it is the raw logs: https://github.com/tylerjw/moveit/runs/2477750664

You can see that the job passes when it should have failed (there were a bunch of failed tests).

ipa-mdl, 2021-05-01:
I proposed a work-around for `ros_buildfarm` (https://github.com/ros-infrastructure/ros_buildfarm/pull/875) as a mitigation while colcon/colcon-test-result#36 is still pending.

There is not much that we can do from our side.
A temporary solution would be to set `BUILDER=catkin_make_isolated` in your prererelase job.

tylerjw, 2021-05-02:
Thank you for looking into this and finding a workaround.  I'm updating all our pre-release jobs to use `catkin_make_isolated` until this is fixed.

PR #694 (2021-07-22): Make catkin_tools the default builder for ROS1 noetic
While I would like to use colcon for ROS1 as well, it has several issues.
I hope that this will not break too many builds.
`BUILDER=colcon` is still supported, but now as an opt-in.

closes #629 and closes #666

@wxmerkt: Please test, if this resolves your build issues.
closed 2021-07-22 by ipa-mdl