ros-industrial/industrial_ci #667: [question] ccache in pre-release

tylerjw, 2021-05-02:
In our testing it seemed like pre-release wasn't using ccache but then I read the code for the pre-release test and I found these lines:

https://github.com/ros-industrial/industrial_ci/blob/e51bbf479becc97a9cadbff9d6d3b9f3374821f8/industrial_ci/src/tests/ros_prerelease.sh#L31-L33

These don't reference the CCACHE_DIR but instead check for the existance of a specific directory.  Does this work if I have the ccache directory somewhere else?

ipa-mdl, 2021-05-02:
This was only a work-around for https://github.com/ros-infrastructure/ros_buildfarm/issues/695 and might not be needed anymore.

> Does this work if I have the ccache directory somewhere else?

I am not sure if ros_buildfarm supports this (https://github.com/ros-infrastructure/ros_buildfarm/pull/582 wasn't merged yet).

ipa-mdl, 2021-05-02:
At the moment `$WORKSPACE` is a temporary directroy, but we could put it into `$BASEDIR` as well.
Then you could cache it from there..

tylerjw, 2021-05-03:
Maybe `WORKSPACE` could be set to be `BASEDIR` if it was defined and then we could cache the ccache directory in there?

ipa-mdl, 2021-05-03:
Actually, `CCACHE_DIR` should work..

https://github.com/ros-industrial/industrial_ci/blob/bfc68c8a3d5adf12ea4074c2149582398b5b6b54/industrial_ci/src/tests/ros_prerelease.sh#L93-L96

ipa-mdl, 2021-05-04:
The ccache is forwarded properly, but it looks like `ros_buildfarm` does not use it anymore (https://github.com/ros-infrastructure/ros_buildfarm/pull/844).

ipa-mdl, 2021-05-04:
See https://github.com/ros-infrastructure/ros_buildfarm/pull/876 as well

ipa-mdl, 2021-05-06:
@tylerjw: please test #670. It should work with `CCACHE_DIR`. `BASEDIR` is now supported as well.