ros-industrial/industrial_ci #668: UNDERLAY feature causes `build_upstream_workspace` to fail to find catkin CMake module

JStech, 2021-05-03:
We attempted to use the UNDERLAY feature in MoveIt Calibration. It seemed to work [initially](https://github.com/ros-planning/moveit_calibration/actions/runs/683966589), but recently every CI run was failing in the `build_upstream_workspace` step with the error `The catkin CMake module was not found,...` (for instance, [here](https://github.com/ros-planning/moveit_calibration/runs/2387071599)). Reverting the use of the UNDERLAY feature fixed this error.

fmessmer, 2021-05-03:
I guess, I intended UNDERLAY to be used **instead** of UPSTREAM_WORKSPACE in https://github.com/ros-industrial/industrial_ci/pull/609

I haven't looked into how both could be made to work together...

gavanderhoorn, 2021-05-03:
It seems something isn't entirely correct in the environment setup:

> WARNING: Your workspace is not extending any other result space, but it is set to use a `linked` devel space layout. This requires the `catkin` CMake package in your source space in order to be built.

JStech, 2021-05-03:
Tagging @tylerjw because he's much more familiar with MoveIt's use of `industrial_ci` than I am.

ipa-mdl, 2021-05-03:
> I haven't looked into how both could be made to work together...

This should not be a problem..

> It seems something isn't entirely correct in the environment setup:

I had similar issues in some tests, but I don't recall why I got this error message.

ipa-mdl, 2021-05-03:
@JStech: `UNDERLAY: /root/ws_moveit` does not point to the install (or devel) space.

https://github.com/ipa-mdl/moveit_calibration/tree/test/underlay seems to work.

The documentation needs to be improved and we should check if `setup.bash` is present.
Another option would be to check `$UNDERLAY/install` as well, but I would prefer the explicit configuration.

@fmessmer, @JStech , @tylerjw: What do you think?

tylerjw, 2021-05-03:
Ah, I misunderstood how to use this feature.  @JStech if you set it to `UNDERLAY: /root/ws_moveit/install` as @ipa-hsd did in his branch that should work.

closed 2021-05-04 by ipa-mdl