ros-industrial/industrial_ci #679: GHA hangs forever with DOWNSTREAM_WORKSPACE

artivis, 2021-05-23:
Hi,

I've successfully set up a Github Actions for a repo of mine (thanks for making it so easy).

Now I've been trying the set up a second job with a `DOWNSTREAM_WORKSPACE` in order to test a few packages that depend on the repo at hand. The issue is that this second job hangs (seemingly) forever with no logs in the GHA window. So far this happened 5/5 times. After cancelling the job, I could inspect the logs and find out that it gets stuck in the `setup_downstream_workspace` tasks at 

```terminal
Processing triggers for libc-bin (2.27-3ubuntu1.4) ...
```

That is, fairly early in the task.

A quick googling seem to suggests that there is indeed some issue(s) with `libc-bin`, often in the context of WSL/Docker. However this deb seem to be installed several times over the course of the whole pipeline so I couldn't say why it steadily fails there.

The PR adding the new job, https://github.com/artivis/laser_odometry/pull/34.

The GHA config file, [ci.yml](https://github.com/artivis/laser_odometry/blob/d0a651d7cce2b7142cfe649061e42061575fb897/.github/workflows/ci.yml#L28).

A failed run can be found [here](https://github.com/artivis/laser_odometry/runs/2651186554?check_suite_focus=true).

ipa-mdl, 2021-05-27:
It seems like one dependency of the downstream workspace needs some interactive configuration, it does not have to be `libc-bin`.
I think we have to improve the overall handling, but I am not sure what exactly needs to be done.

ipa-mdl, 2021-07-17:
One of your downstream dependencies seems to be a private repository.
Locally I see the prompt 
```
Username for 'https://github.com': .......
```
You could setup an access token or you could checkout your private repository with the checkout action and then use the local folder as part of your downstream workspace. Or even simpler: do not test private downstream dependencies, if this is an option.

ipa-mdl, 2021-07-17:
Ah, no, you just have a typo in https://github.com/artivis/laser_odometry/blob/d0a651d7cce2b7142cfe649061e42061575fb897/.downstream_ci.rosinstall#L3

closed 2021-07-17 by ipa-mdl

artivis, 2021-07-17:
> Ah, no, you just have a typo in https://github.com/artivis/laser_odometry/blob/d0a651d7cce2b7142cfe649061e42061575fb897/.downstream_ci.rosinstall#L3

That was it 😅. My bad, thanks!