ros-industrial/industrial_ci #693: LD_LIBRARY_PATH: unbound variable

MatthijsBurgh, 2021-06-08:
I get the following error during `abi_dump_target`:
```
 /home/runner/work/_actions/ros-industrial/industrial_ci/master/industrial_ci/src/tests/abi_check.sh: line 65: LD_LIBRARY_PATH: unbound variable
```
(https://github.com/orocos/orocos_kinematics_dynamics/runs/2774068469?check_suite_focus=true#step:3:808)

This is the run after rebasing the PR. The pull_request run of the PR ran without any issues: https://github.com/orocos/orocos_kinematics_dynamics/runs/2773778532?check_suite_focus=true

ipa-mdl, 2021-06-08:
Thanks for reporting!

ipa-mdl, 2021-06-08:
I wonder why `LD_LIBRARY_PATH` is not set..
Just fixing the "unbound variable" will not help, because the value is needed for the ABI dumper.

I will try to investigate this further using your source repo.

MatthijsBurgh, 2021-07-10:
@ipa-mdl Have you been able to find the cause and/or a solution?

ipa-mdl, 2021-07-16:
@MatthijsBurgh: Not yet.. could you create a branch with the non-rebased version. I will test it locally then.

ipa-mdl, 2021-07-16:
> could you create a branch with the non-rebased version. I will test it locally then.

I managed to clone https://github.com/orocos/orocos_kinematics_dynamics/commit/84f04d00d82107474ab619ccd211da913a520f27

ipa-mdl, 2021-07-16:
At a [closer look at your failing job](https://github.com/orocos/orocos_kinematics_dynamics/runs/2774068469?check_suite_focus=true#step:3:804) your base repository seems to be empty.

It looks like something is wrong with the checkout action. You might have to update your `if` clause.

ipa-mdl, 2021-07-16:
If I run it locally:

```
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
abi_check_liborocos-kdl
Preparing, please wait ...
Comparing ABIs ...
Comparing APIs ...
Creating compatibility report ...
Binary compatibility: 100%
Source compatibility: 100%
Total binary compatibility problems: 0, warnings: 0
Total source compatibility problems: 0, warnings: 0
Report: /root/abicheck/release-1.5/liborocos-kdl.html
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
'abi_check_liborocos-kdl' returned with code '0' after 0 min 0 sec
```

It does not seem to an issue with `industrial_ci`, so I will close this for now.

closed 2021-07-16 by ipa-mdl

MatthijsBurgh, 2021-07-16:
@ipa-mdl Thanks for your efforts. I see that I didn't update the `if` conditions for the checkout action. So thanks for your suggestion.