ros-industrial/industrial_ci #697: pre-release tests that more closely resemble what the buildfarm does for ros2

tylerjw, 2021-07-01:
We discovered recently is that the pre-release tests in industrial_ci use colcon but the official buildfarm does not.  This lead to builds passing in our pre-release tests and then failing on the buildfarm.

Quoting from here: https://github.com/ros-planning/moveit2/pull/532

> The issue is that ${PROJECT_NAME}_VERSION is not set automatically with colcon, it needs to be set explicitly using ament_package_xml(). Since the sonames are configured in all packages by setting the VERSION property, the actual file name ends up ending with a .. With this fix the sonames are set to the corresponding package versions for all libraries. In the future we should consider setting up explicit ABI versions.

If the pre-release test used `debuild` like the official buildfarm instead of colon it would have caught this issue.  I'm not sure how difficult it would be to implement that change.  Tagging @jspricke @henningkayser as they fixed the issue for our moveit2 release.

jspricke, 2021-07-01:
Note that you would also need to order the packages in one repo by dependency, run bloom, create a Debian source package and a minimal build environment (sbuild, pbuilder) on each. Not sure if it makes sense to do it in this project.

ipa-mdl, 2021-07-01:
> We discovered recently is that the pre-release tests in industrial_ci use colcon but the official buildfarm does not. This lead to builds passing in our pre-release tests and then failing on the buildfarm.

The `industrial_ci` pre-release test runs the pre-release test of `ros_buildfarm` (which uses colcon under the hood).
We just have some extra steps to setup docker for the different scenarios, make sure that it is not run as root and preparing `UPSTREAM_WORKSPACE` and `DOWNSTREAM_WORKSPACE`.

tylerjw, 2021-07-01:
> The `industrial_ci` pre-release test runs the pre-release test of `ros_buildfarm`

So maybe I should make this issue against the ros_buildfarm so hopefully they can have the official pre-release process more closely resemble the buildfarm?

jspricke, 2021-07-08:
I've written a simple script to do prerelease testing similar to the ROS buildfarm (i.e. without colcon), maybe you can integrate that:
```
#!/bin/sh -e
# run pre release tests for a repo or workspace
# setup sbuild: https://wiki.ubuntu.com/SimpleSbuild
# sudo apt install python3-colcon-package-information python3-bloom

REPO=$(mktemp -d)

# alternatively catkin_topological_order
for PKG_PATH in $(colcon list -tp); do
  echo "Building $PKG_PATH"
  (
  cd "$PKG_PATH"
  # debian-inc=2 in case there are packages published from the build farm already
  bloom-generate rosdebian --ros-distro=galactic --debian-inc=2
  # https://github.com/ros-infrastructure/bloom/pull/643
  echo 11 > debian/compat
  # ros2-testing is only needed if there are unpublished dependencies
  # dpkg-source-opts: no need for upstream.tar.gz
  sbuild --extra-repository="deb http://packages.ros.org/ros2/ubuntu focal main" \
         --extra-repository="deb http://packages.ros.org/ros2-testing/ubuntu focal main" \
         --extra-repository-key=/usr/share/keyrings/ros-archive-keyring.gpg --no-run-lintian \
         --extra-package="$REPO"  --dpkg-source-opts="-Zgzip -z1 --format=1.0 -sn"
  # https://bugs.debian.org/990734
  mv ../*.deb "$REPO/"
  )

done
git clean -dfx
rm -r "$REPO"
```

ipa-mdl, 2021-07-09:
@jspricke: I am having a hard time setting up sbuild in Docker. Are there any alternatives?

jspricke, 2021-07-09:
There is this list:
https://wiki.debian.org/SystemBuildTools#Package_build_tools
But it could be that you have to work around some missing features.
Why would you use Docker? You are probably missing the chroot and bind mount syscalls.

ipa-mdl, 2021-07-21:
> Why would you use Docker? You are probably missing the chroot and bind mount syscalls.

`industrial_ci` is built around Docker. Shell "isolation" has been added recently, but assumes that it is run in a disposable environment like Github actions (Docker-based as well).

jspricke, 2021-07-22:
* Mathias Lüdtke ***@***.***> [2021-07-21 14:12]:
>> Why would you use Docker? You are probably missing the chroot and bind mount syscalls.
>
>`industrial_ci` is built around Docker. Shell "isolation" has been added recently, but assumes that it is run in a disposable environment like Github actions (Docker-based as well).

As far as I found, chroot and bind mounts would need a privileged Docker 
container and that is not available in Github actions.
You could also try the unshare chroot mode of sbuild but that is still 
experimental. 
In addition to the wiki link above, Salsa CI could serve as an 
inspiration as well.

ipa-mdl, 2021-07-22:
I came up with a simpler approach (https://github.com/ros-industrial/industrial_ci/pull/719), which just runs `bloom-generate`, `dpkg-buildpackage` and then `dpkg -i` for each package in topological order.

@tylerjw: Please test
@jspricke: I am not using `debian/compat=11` yet. Would it use all cores by default or to I have to configure something else as well?

jspricke, 2021-07-22:
#719 is not a solution to this issue, see my comments.
Regarding parallel builds, you can add `--parallel` to the `dh` in `debian/rules` but `compat=11` does basically the same plus other benefits.

ipa-mdl, 2021-08-02:
`TEST=debians` (#719) should be much closer to what the buildfarm does during **release**.
It is a little bit slow (1,5h for moveit2), but this due to the complete isolation of the builds.
It should be possible to speed it up with `ccache`,  but did have not looked into that yet.

closed 2021-08-02 by ipa-mdl