ros-industrial/industrial_ci #723: DOWNSTREAM_WORKSPACE cannot resolve github.com

fmessmer, 2021-07-22:
I'm stuck during setup_downstream_workspace with the fillowing output:
```
 At least one hostname (github.com) is unknown, switching to a single worker to allow interactively answering the ssh question to confirm the fingerprint
  
  The authenticity of host 'github.com (140.82.121.4)' can't be established.
  RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
  Error: The operation was canceled.
```

where the github action gets cancelled after 1h
I use the same rosinstall file with private github repos for DOWNSTREAM_WORKSPACE that works just fine for UPSTREAM_WORKSPACE
I dont really see what's wrong...any idea?

ipa-mdl, 2021-07-22:
Are you using a self-hosted runner?
Looks like you are using a git URL for DOWNSTREAM_WORKSPACE, but not for UPSTREAM_WORKSPACE.

ipa-mdl, 2021-07-22:
See https://github.com/ros-industrial/industrial_ci/issues/574 as well.

fmessmer, 2021-07-23:
ah, right.....thanks!
I was missing the line `AFTER_INIT` from https://github.com/ros-industrial/industrial_ci/issues/574#issuecomment-766706003

the other rosinstall file was not used by the github action and confused me
sorry for the noise

closed 2021-07-23 by fmessmer