ros-industrial/industrial_ci #742: Problem with running industrial_ci locally with act

destogl, 2021-09-20:
Hi,

I am trying to run industrial_ci locally using [act](https://github.com/nektos/act) (also mentioned in #474). I had to do some small changes to the workflow file (I will report them later when I have all running), but still having an error which I am not sure why is happening and what is actually environment it is executed in. The same workflow file is working as GH Action.

The error-output is following:

```
[Binary Build/rolling binary build] Exec command '[bash --noprofile --norc -e -o pipefail /<ROS_PACKAGE_WITH_WORKFLOW>/workflow/composite-0.sh]'
| Running test 'source_tests'
| 
[Binary Build/rolling binary build]   ❓  ##[group]pull_docker_image
| 
| $ docker pull ubuntu:focal
| focal: Pulling from library/ubuntu
| Digest: sha256:9d6a8699fb5c9c39cf08a0871bd6219f0400981c570894cd8cbea30d3424a31f
| Status: Image is up to date for ubuntu:focal
| docker.io/library/ubuntu:focal
| ##[endgroup]
| 'pull_docker_image' returned with code '0' after 0 min 1 sec
| 
| $ docker create --init --env-file /var/run/act/actions/ros-industrial-industrial_ci@master/industrial_ci/src/isolation/docker.env --rm -v /<ROS_PACKAGE_WITH_WORKFLOW>:/<ROS_PACKAGE_WITH_WORKFLOW>:ro -e TARGET_REPO_PATH=/<ROS_PACKAGE_WITH_WORKFLOW> -v /run/act/actions/ros-industrial-industrial_ci@master/industrial_ci/src:/run/act/actions/ros-industrial-industrial_ci@master/industrial_ci/src:ro -e ICI_SRC_PATH=/run/act/actions/ros-industrial-industrial_ci@master/industrial_ci/src -v /<ROS_PACKAGE_WITH_WORKFLOW>/.work:/<ROS_PACKAGE_WITH_WORKFLOW>/.work:rw -e BASEDIR=/<ROS_PACKAGE_WITH_WORKFLOW>/.work -v /<ROS_PACKAGE_WITH_WORKFLOW>/.ccache:/<ROS_PACKAGE_WITH_WORKFLOW>/.ccache:rw -e CCACHE_DIR=/<ROS_PACKAGE_WITH_WORKFLOW>/.ccache -t --entrypoint  -w /<ROS_PACKAGE_WITH_WORKFLOW> ubuntu:focal /bin/bash /var/run/act/actions/ros-industrial-industrial_ci@master/industrial_ci/src/run.sh source_tests run_source_tests
| Copy credentials: /home/runner/.ssh
| 
| $ docker start -a 4f68b174eecb9e4cb2af4d5a6e422d01ac84c8d012ba082d24a68e48549f247e
| /bin/bash: /var/run/act/actions/ros-industrial-industrial_ci@master/industrial_ci/src/run.sh: No such file or directory
DEBU[0004] exit with `FAILURE`: 127                     
[Binary Build/rolling binary build]   ❌  Failure - ros-industrial/industrial_ci@master
```

gavanderhoorn, 2021-09-20:
PR #582 already reports "problems with mounting some folders", which seems like it could be the cause of what you are seeing as well.

ipa-mdl, 2021-09-20:
> which seems like it could be the cause of what you are seeing as well.

I would second that.

Docker mounts are always tricky, especially if mounted folder gets passed through.
To keep it simple the `industrial_ci` just uses the same names inside and outside, which works well for CI, DinD and Docker socket sharing.

Not sure how act is solving this issue.

ipa-mdl, 2021-09-20:
Update: I just tried it and it worked on my system (WSL2 & Docker Desktop).

I really just downloaded the [latest version](https://github.com/nektos/act/releases) and ran ```./act -j builders```.

@destogl: Have you trief running act in privileged  mode? or with `-b`?

ipa-mdl, 2021-09-20:
> The concrete issue is that ICI_SRC_PATH exists, but it is completely empty. Also is empty on my local machine but first when docker start in this line is executed.

As as I recall it this happens whenever the path does not exist on the Docker host.
I will test it as well under Ubuntu 18.04.

> Maybe I did something wrong in docker configuration?

Which version are you using?

ipa-mdl, 2021-09-20:
> I will test it as well under Ubuntu 18.04.

Works fine (for `-j builders`).

destogl, 2021-09-21:
> Which version are you using?

Docker version 20.10.8, build 3967b7d

Ok I tested this repository and it work for me to. The issue I have is with other repositories using `industrial_ci` in GH Actions for CI. For example [`ros-controls/ros2_control`](https://github.com/destogl/ros2_control)

Here exactly what I am doing:
- clone ros2_control@master
- execute `act -j ci_binary` in the ros2_control folder
  This returns error. This is probably something that `act` does not properly parses the parameters in the action file. Therefore I chnage the `env` part to explicity set `ROS_DISTRO` and `ROS_REPO` variables
  ````
  env: #${{matrix.env}}
          ROS_DISTRO: galactic
          ROS_REPO: main
  ```
  This is not so huge issue for now, it's just not user-friendly.

- execute again `act -j ci_binary` in the ros2_control folder
  How the "main" error happens: (if testing watch out on the output, there are two jobs/cointainers started)
  ```
  | Copy credentials: /root/.ssh
  | 
  | $ docker start -a cc184fe2060ad6145c0e54271ab690011b77dde892aa29f0d44a6996c6106c6a
  | /bin/bash: /var/run/act/actions/ros-industrial-industrial_ci@master/industrial_ci/src/run.sh: No such file or directory
  [Build & Test ros2_control with binary install/Galactic binary job-1]   ❌  Failure - ros-industrial/industrial_ci@master
  ```

- Following your pointers I fould out that the path `/var/run` is symlink to `/run`
- For debugging to resolve this I am resolving the real path in the `main_ci.sh` (see [here](https://github.com/destogl/industrial_ci/blob/71b6e1fd832da9074cf2bf1cc555184b7f6aca49/industrial_ci/src/ci_main.sh#L74-L76)
- I added some additional output to try to understand what is happening I added some more output into my fork at `act-tests` branch

Here the output of my test with additional prints:
```
| Running test 'source_tests'
| With link path: /var/run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src
| Absolute path: /run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src
| 
[Build & Test ros2_control with binary install/Galactic binary job]   ❓  ##[group]pull_docker_image
| 
| $ docker pull ubuntu:focal
| focal: Pulling from library/ubuntu
| Digest: sha256:9d6a8699fb5c9c39cf08a0871bd6219f0400981c570894cd8cbea30d3424a31f
| Status: Image is up to date for ubuntu:focal
| docker.io/library/ubuntu:focal
| ##[endgroup]
| 'pull_docker_image' returned with code '0' after 0 min 1 sec
| 
| $ docker create --init --env-file /run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src/isolation/docker.env --rm -v <ROS2_WS>/ros2_control:<ROS2_WS>/ros2_control:ro -e TARGET_REPO_PATH=<ROS2_WS>/ros2_control -v /run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src:/run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src:ro -e ICI_SRC_PATH=/run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src -t --entrypoint  -w <ROS2_WS>/ros2_control ubuntu:focal /bin/ls -al /run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src
| Copy credentials: /root/.ssh
| Just before docker start command: /run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src
| total 88
| drwxr-xr-x 7 root root  4096 Sep 21 10:32 .
| drwxr-xr-x 6 root root  4096 Sep 21 10:32 ..
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 builders
| -rw-r--r-- 1 1000 1000  2458 Sep 21 10:32 ci_main.sh
| -rw-r--r-- 1 1000 1000  4122 Sep 21 10:32 deprecated.sh
| -rw-r--r-- 1 1000 1000  2823 Sep 21 10:32 env.sh
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 folding
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 isolation
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 keys
| -rw-r--r-- 1 1000 1000  5479 Sep 21 10:32 ros.sh
| -rwxr-xr-x 1 1000 1000  2252 Sep 21 10:32 run.sh
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 tests
| -rw-r--r-- 1 1000 1000 14142 Sep 21 10:32 util.sh
| -rw-r--r-- 1 1000 1000 13993 Sep 21 10:32 workspace.sh
| 
| $ docker start -a 42bed32274e92a34ac05a76c9fd38b9f92e2868112d705106aede090a31fe3c3
| total 4
| drwxr-xr-x 2 root root   40 Sep 21 09:50 .
| drwxr-xr-x 3 root root 4096 Sep 21 10:32 ..
| Just after docker start command: /run/act/actions/destogl-industrial_ci@act-tests/industrial_ci/src
| total 88
| drwxr-xr-x 7 root root  4096 Sep 21 10:32 .
| drwxr-xr-x 6 root root  4096 Sep 21 10:32 ..
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 builders
| -rw-r--r-- 1 1000 1000  2458 Sep 21 10:32 ci_main.sh
| -rw-r--r-- 1 1000 1000  4122 Sep 21 10:32 deprecated.sh
| -rw-r--r-- 1 1000 1000  2823 Sep 21 10:32 env.sh
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 folding
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 isolation
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 keys
| -rw-r--r-- 1 1000 1000  5479 Sep 21 10:32 ros.sh
| -rwxr-xr-x 1 1000 1000  2252 Sep 21 10:32 run.sh
| drwxr-xr-x 2 root root  4096 Sep 21 10:32 tests
| -rw-r--r-- 1 1000 1000 14142 Sep 21 10:32 util.sh
| -rw-r--r-- 1 1000 1000 13993 Sep 21 10:32 workspace.sh
[Build & Test ros2_control with binary install/Galactic binary job]   ✅  Success - destogl/industrial_ci@act-tests
```

ipa-mdl, 2021-09-21:
> This returns error. This is probably something that act does not properly parses the parameters in the action file. Therefore I chnage the env part to explicity set ROS_DISTRO and ROS_REPO variables

It is not `act`'s fault..
This just a loop hole that we used in the initial version of our action templates, but it is not really supported (https://github.com/ros-industrial/industrial_ci/issues/590).

ipa-mdl, 2021-09-21:
> Ok I tested this repository and it work for me to. The issue I have is with other repositories using industrial_ci in GH Actions for CI. 

On normal runners Github treats `uses` the same for local paths and repo paths.
The `industrial_ci` jobs uses './.industrial_ci/' to test new changes.
If I use a local copy, then it works for ros2_control as well.
I guess we need some change in https://github.com/ros-industrial/industrial_ci/blob/master/.github/action.sh

destogl, 2021-09-21:
> I guess we need some change in https://github.com/ros-industrial/industrial_ci/blob/master/.github/action.sh

Do you have any idea what should be changed? I could probably test something, but any pointer would be great!

> On normal runners Github treats uses the same for local paths and repo paths.
> The industrial_ci jobs uses './.industrial_ci/' to test new changes.

The question is actually, why the `/run/...` path is not mapped properly into internal docker. I am not sure how local path "`./.industrial_ci/` would influence/solve this.

ipa-mdl, 2021-09-21:
Okay, tested it a little bit more..

It seems to work for `ISOLATION: shell`.

Apparently, `act` is not sharing `/run/act` or `/var/run/act` with the host, but sets these up in the container.
So it should work, if ´ICI_SRC_PATH` is not bind-mounted, but copied into the container.

ipa-mdl, 2021-09-21:
> So it should work, if ICI_SRC_PATH is not bind-mounted, but copied into the container.

@destogl: Please try https://github.com/ipa-mdl/industrial_ci/tree/feature/act-support

```yaml
      - uses: ipa-mdl/industrial_ci@feature/act-support
````

ipa-mdl, 2021-09-21:
https://github.com/nektos/act/pull/776 might solve the issue as well

ipa-mdl, 2021-11-12:
Merged #743. I think this can be closed now.

closed 2022-03-17 by destogl