ros-industrial/industrial_ci #744: Sources list is modified even when DOCKER_IMAGE is passed.

InigoMoreno, 2021-09-22:
In the docs, it says that Setting `DOCKER_IMAGE` disables the set-up of `ROS` based on `ROS_REPO` ([ref](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#pulling-docker-image-from-an-online-hub)).

However, this is not true. See how in this minimal example, the `/etc/apt/sources` is changed even though `DOCKER_IMAGE` is provided:
```bash
$ rosrun industrial_ci run_ci DOCKER_IMAGE=osrf/ros:melodic-desktop \
                              ROS_REPO=testing \
                              BEFORE_INIT="cat /etc/apt/sources.list.d/ros*" \
                              AFTER_INIT="cat /etc/apt/sources.list.d/ros*; exit 1"

before_init
deb http://packages.ros.org/ros/ubuntu bionic main

after_init
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros-testing/ubuntu bionic main

```

This is probably because of the code in these lines:
https://github.com/ros-industrial/industrial_ci/blob/c43b9a44265d8b498077e36edd53a45983a3b105/industrial_ci/src/workspace.sh#L143-L150

ipa-mdl, 2021-09-22:
Thanks for reporting. The documentation on this a little bit outdated.
If you pass `ROS_REPO`, the code you pointed to will check for it and update it, if necessary.
In your case `osrf/ros:melodic-desktop` uses the main repo, so it will get updated.

InigoMoreno, 2021-09-23:
This also happens even if you don't pass `ROS_REPO`, as `testing` is the default value.

IMO, this is more than a documentation issue. industrial_ci should not touch the sources if its using a `DOCKER_IMAGE`.

ipa-mdl, 2021-10-06:
> This also happens even if you don't pass ROS_REPO, as testing is the default value.

Ah yes, this is somehow unintended.

> industrial_ci should not touch the sources if its using a DOCKER_IMAGE

Then it would not work with arbitrary images.
I could imagine making `ROS_REPO` mandatory in this case.

ipa-mdl, 2021-10-06:
#748 offers an opt-out

I did not find any good solution yet.
One option is to not touch the set-up, if `ROS_REPO` is not set and any "ros*.list" is found.
However, this would change the behavior for users, which set `DOCKER_IMAGE`, but not `ROS_REPO`.
I guess we will have to raise an error in this case.
@InigoMoreno: Any other ideas?

ipa-mdl, 2021-10-06:
> I would argue that, as the documentation states that setting DOCKER_IMAGE disables the setup of ROS based on ROS_REPO

The documentation does not reflect the updates in the master branch, but describes the legacy version.

`ROS_DISTRO` and `ROS_REPO` are not ignored anymore!

If you set `DOCKER_IMAGE=ros:melodic` and `ROS_DISTRO=noetic` the build will fail, which is the correct behavior.
The issue is that `ROS_REPO=testing` will be enforced, even if this is not intended.

Right now I am struggling to define a heuristic that will detect this case properly.
Please note that this issue exists as well for `ISOLATION=shell` (without `DOCKER_IMAGE`)

PR #748 (2021-11-11): Refactor ROS_REPO handling
`ROS_REPO` should not overwrite an existing ROS setup by default (fixes #744)
If `ROS_REPO` does no match the existing ROS setup, a warning will be raised.
closed 2021-11-11 by ipa-mdl