ros-industrial/industrial_ci #745: ROS2 launch-tests fails if files need to be installed

agutenkunst, 2021-09-23:
Maybe this is related to https://github.com/ros-industrial/industrial_ci/issues/738

Context:
I have an launch-test (.test.py) which needs a unittest-based python file.

Inside my CMakeLists.txt I setup this test using:

```
  install(FILES
    test/integrationtest_unittest.py
    DESTINATION ${CMAKE_CURRENT_BINARY_DIR}
    PERMISSIONS OWNER_READ OWNER_EXECUTE
  )
  add_ros_test(test/integrationtest_unittest.test.py ARGS "test_binary_dir:=${CMAKE_CURRENT_BINARY_DIR}")
```
the relevant files are [integrationtest_unittest.py](https://github.com/agutenkunst/ros2_testing_demo_pkg/blob/4b94613ff28aa04216b6c37b8efe5cc3c36ce1fe/test/integrationtest_unittest.py) and [integrationtest_unittest.test.py](https://github.com/agutenkunst/ros2_testing_demo_pkg/blob/feature/integrationtest-unittest/test/integrationtest_unittest.test.py)
this works locally using `colcon build && colcon test && colcon test-result` but on github action using industrial_ci I get:

```
 FileNotFoundError: [Errno 2] No such file or directory: '/root/target_ws/build/ros2_testing_demo_pkg/integrationtest_unittest.py'
```
Full log: https://github.com/agutenkunst/ros2_testing_demo_pkg/runs/3683139782

I investigated a bit and checked with `BEFORE_RUN_TARGET_TEST: "tree -L 2 /root/target_ws/build/ && ls -la /root/target_ws/build/ros2_testing_demo_pkg/integrationtest_unittest.py"` and it seems that the file exists in the correct place.

The whole package is located at https://github.com/agutenkunst/ros2_testing_demo_pkg

ipa-mdl, 2021-10-05:
> this works locally using colcon build && colcon test && colcon test-result

Did you source your setup.bash beforehand?

agutenkunst, 2021-10-06:
No I didn't source setup.bash.
If I run
```
colcon build && colcon test && colcon test-result --verbose --all
```
I get 
```
Starting >>> ros2_testing_demo_pkg
Finished <<< ros2_testing_demo_pkg [6.00s]                     

Summary: 1 package finished [6.12s]
Starting >>> ros2_testing_demo_pkg
Finished <<< ros2_testing_demo_pkg [1.49s]          

Summary: 1 package finished [1.66s]
build/ros2_testing_demo_pkg/Testing/20211006-0444/Test.xml: 3 tests, 0 errors, 0 failures, 0 skipped
build/ros2_testing_demo_pkg/test_results/ros2_testing_demo_pkg/test_integrationtest_gtest.test.py.xunit.xml: 1 test, 0 errors, 0 failures, 0 skipped
build/ros2_testing_demo_pkg/test_results/ros2_testing_demo_pkg/test_integrationtest_unittest.test.py.xunit.xml: 1 test, 0 errors, 0 failures, 0 skipped
build/ros2_testing_demo_pkg/test_results/ros2_testing_demo_pkg/unittest_gtest.gtest.xml: 1 test, 0 errors, 0 failures, 0 skipped

Summary: 6 tests, 0 errors, 0 failures, 0 skipped
```
when running with https://github.com/agutenkunst/ros2_testing_demo_pkg/tree/feature/integrationtest-unittest

agutenkunst, 2021-10-06:
#### Update
I had to do some changes to the test but it did not influence the original issue of
```
FileNotFoundError: [Errno 2] No such file or directory: '/root/target_ws/build/ros2_testing_demo_pkg/integrationtest_unittest.py'
``` 
see https://github.com/agutenkunst/ros2_testing_demo_pkg/runs/3811973464

ipa-mdl, 2021-10-06:
I get the same errors with `run_ci`..
Why are you installing the test into the build space?
I guess it will fail locally with colcon as well, if you clear the build space first.