ros-industrial/industrial_ci #758: CI Pipeline breaks when configured to run linters [black]

nocoinman, 2021-11-08:
Pipelines with `BLACK_CHECK` set to `true` fails with message
>   E: The repository 'http://packages.ros.org/ros/ubuntu bullseye Release' does not have a Release file.

Workflow config file
---

```yaml
name: lint

# Controls when the action will run. Triggers the workflow on push or pull request
on:
  - push
  - pull_request

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - ROS_DISTRO: noetic
            ROS_REPO: testing
            CLANG_FORMAT_CHECK: file
            CLANG_FORMAT_VERSION: 11

          - ROS_DISTRO: noetic
            ROS_REPO: testing
            BLACK_CHECK: true

    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{matrix.env}}
```

Output on failure
---

```
 $ sudo sed -i /^# deb.*multiverse/ s/^# // /etc/apt/sources.list
  
  $ sudo apt-get update -qq
  
  $ sudo apt-get -qq install -y --no-upgrade --no-install-recommends apt-utils build-essential gnupg2 dirmngr ca-certificates | grep -E 'Setting up' 
  Setting up gnupg2 (2.2.27-2) ...
  Setting up apt-utils (2.2.4) ...
  Setting up build-essential (12.9) ...
  
  $ sudo apt-get -qq install -y --no-upgrade --no-install-recommends lsb-release | grep -E 'Setting up' 
  Setting up distro-info-data (0.51) ...
  Setting up lsb-release (11.1.0) ...
  /usr/share/keyrings/ros-archive-keyring.gpg
  -------------------------------------------
  pub   rsa4096 2019-05-30 [SC] [expires: 2025-06-01]
        C1CF 6E31 E6BA DE88 68B1  72B4 F42E D6FB AB17 C654
  uid           [ unknown] Open Robotics <info@osrfoundation.org>
  
  
  $ sudo apt-get update -qq
  E: The repository 'http://packages.ros.org/ros/ubuntu bullseye Release' does not have a Release file.
  'sudo apt-get update -qq' returned with 100
'init' returned with code '100' after 0 min 8 sec
```

ipa-mdl, 2021-11-08:
Thanks for reporting!
Normally, ROS should not get set-up for `BLACK_CHECK `.
It might be that some of the recent change broke it.

nocoinman, 2021-11-08:
@ipa-mdl 
`ROS_DISTRO` seems to be a mandatory variable.
I first tried with the following configuration

```yaml
jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
           - CLANG_FORMAT_CHECK: file
             CLANG_FORMAT_VERSION: 11

           - BLACK_CHECK: true
```

and it failed with the message

```
Please set ROS_DISTRO, OS_CODE_NAME or DOCKER_IMAGE.
```

So then I tried with this

```yaml
jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - OS_CODE_NAME: bionic
            CLANG_FORMAT_CHECK: file
            CLANG_FORMAT_VERSION: 11
            BLACK_CHECK: true
```

and now it fails with the message

```
ROS_DISTRO is not set
```

ipa-mdl, 2021-11-08:
@nocoinman: Did it work before?

ipa-mdl, 2021-11-08:
> ROS_DISTRO seems to be a mandatory variable.

It is mandatory for `CLANG_FORMAT_CHECK` (not sure, if it can be made optional).
For `BLACK_CHECK` is is not needed and apparently it fails, if it is provided.

nocoinman, 2021-11-08:
> @nocoinman: Did it work before?

The config file mentioned in the issue used to work in one of my older projects, like a few months back. I just copied the file to my current project and now it fails.

nocoinman, 2021-11-08:
> > ROS_DISTRO seems to be a mandatory variable.
> 
> It is mandatory for `CLANG_FORMAT_CHECK` (not sure, if it can be made optional). For `BLACK_CHECK` is is not needed and apparently it fails, if it is provided.

Great! 

I managed to make the workflow work with the below configuration from what you have said! It does the job now.

```yaml
name: lint

# Controls when the action will run. Triggers the workflow on push or pull request
on:
  - push
  - pull_request

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - ROS_DISTRO: noetic
            CLANG_FORMAT_CHECK: file
            CLANG_FORMAT_VERSION: 11

          - OS_CODE_NAME: bionic
            BLACK_CHECK: true

    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{matrix.env}}
```

Although I am afraid the bug still exists. 

Thanks for your help!

ipa-mdl, 2021-11-08:
> Although I am afraid the bug still exists.

Yes, there is still an issue. `BLACK_CHECK` uses the `python:3` by default and this is not supported by ROS.

ipa-mdl, 2021-11-08:
#759 fixes this issue.

> OS_CODE_NAME: bionic

This will not work, because BLACK_CHECK uses `python:3` by default. You could choose a different OS by passing a different DOCKER_IMAGE or ROS_DISTRO (with the fix).

PR #759 (2021-11-08): Do not set DOCKER_IMAGE in BLACK_CHECK, if ROS_DISTRO is set
closes #758
closed 2021-11-08 by ipa-mdl