ros-industrial/industrial_ci #767: Use deterministic directory for clang_format_check

wxmerkt, 2022-01-06:
Hi,
I want to download a specific `.clang-format` config file with a before/after target hook. I.e., by setting `AFTER_PREPARE_CLANG_FORMAT_CHECK="wget ... -O /clang-format-source-space/.clang-format"`.

This is complicated by the source space being in a temporary directory:
https://github.com/ros-industrial/industrial_ci/blob/008159b3948669bf3583c72c054e1c4f231d1c14/industrial_ci/src/tests/clang_format_check.sh#L25

Would you consider a PR for the following suggestion:
1. Use a deterministic directory/source space, e.g. `/tmp/clang-format-check`
2. Split the set-up phase (install the package etc.) from `run_clang_format_check` to `prepare_clang_format_check`

Or is there an alternative way this could work that I missed?

mathias-luedtke, 2022-01-07:
> Or is there an alternative way this could work that I missed?

No, directly. You could download .clang-format before running `industrial_ci`.
We could export the path as well, so you can use it in the hook.
Or we use the the `current_ws` systems.
Another idea: We could add http(s) support for `CLANG_FORMAT_CHECK`. This way you do not need a hook.

> Use a deterministic directory/source space, e.g. /tmp/clang-format-check

I would prefer something local, e.g. in basedir.

> Split the set-up phase (install the package etc.) from run_clang_format_check to prepare_clang_format_check

Currently there are 3 steps:
* `install_clang_format`
* `prepare_sourcespace`
*  `run_clang_format_check`

What exactly do you want to split?

wxmerkt, 2022-01-07:
I prepared a PR in #768 with a sample solution. I noticed that `install_clang_format_check` didn't exist yet, so I split out some of the code from `run_clang_format_check` into `install_clang_format_check`. I am open for comments on what would be best to implement (the solution in #768 works for our use case but requires the `wget ... ` in the `AFTER_PREPARE_SOURCESPACE` hook.