ros-industrial/industrial_ci #772: RViz randomly crashes in the CI for Foxy

BenjaminHug8, 2022-02-04:
Hi all,

We are running [CI tests](https://github.com/cyberbotics/webots_ros2/blob/1da1f08a928861839c68de7ccf9515a3528b570d/.github/workflows/test.yml#L23) with your tool but with have issues with RViz for certain tests for our [turtlebot3burger example](https://github.com/cyberbotics/webots_ros2/wiki/Example-TurtleBot3-Burger) (these tests are only run with Foxy).

These 2 tests might randomly fail and output the following for RViz:

```
[rviz2-78] loaded library "/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so"
[rviz2-78] qt.qpa.xcb: could not connect to display 
[rviz2-78] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-78] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-78] 
[rviz2-78] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
```

Here is an example of an [unsuccessful test](https://github.com/cyberbotics/webots_ros2/runs/5065553983?check_suite_focus=true#step:3:18691).

Our simulator Webots does also used Qt but as sometimes the tests pass, I have doubt that the issue comes from Qt (I have checked that RViz do not use the Qt libraries coming with Webots).

Is it common that RViz crashes in the CI, or is there any setup I should have to ensure that it won't crash?

Let me know if I can give any other information if needed.

Thanks in advance for your time,
Benjamin

Briancbn, 2022-02-04:
The CI is usually ran in a headless (no GUI) environment, it seems that `xcb` cannot find the display. You can verify it by adding `echo $DISPLAY ` while nothing is returned.

I find this post in the qt forum by `KH-219Design`, not sure if it will help. https://forum.qt.io/topic/133749/adding-gui-tests-to-ci-without-display/9

BenjaminHug8, 2022-02-07:
> The CI is usually run in a headless (no GUI) environment, it seems that `xcb` cannot find the display. You can verify it by adding `echo $DISPLAY ` while nothing is returned.

That is the case, you are right.

> I find this post in the qt forum by `KH-219Design`, not sure if it will help. https://forum.qt.io/topic/133749/adding-gui-tests-to-ci-without-display/9 . Maybe the maintainer will have more insights

Ok thank you I will check that and in case I found anything useful I will write it here for any future users having the same issue as we do.

mathias-luedtke, 2022-02-13:
If the builds fail randomly (-> not always), then you have a race condition somewhere.
And normally, a crash in RVIZ should not make the CI fail, so it might be just a symptom.

BenjaminHug8, 2022-02-14:
Hi @mathias-luedtke, Rviz is just simply installed via `apt install` and the CI does not crash, we are just testing an update of the map in Rviz thus the test fails when Rviz crashes.