ros-industrial/industrial_ci #773: docker and static analysis

muttistefano, 2022-02-04:
Hi everyone,
im trying to integrate a software analysis tool(sonarcloud) into a github action that also includes industrial_ci to build a ROS package.
The problem is that the build is done into the docker, by industrial_ci, and after that i cannot access the build folder wich is needed by sonarcloud.
Is there a way to use industrial_ci without the docker part, or another work-around?
Thanks

Briancbn, 2022-02-04:
Hi @muttistefano, Yes you can, by setting `ISOLATION: "shell"`.

Here is an example, https://github.com/ros-industrial/easy_manipulation_deployment/blob/eu-conference/.github/workflows/build.yml

muttistefano, 2022-02-04:
Hi @Briancbn , thank you, it is working without docker now.
Is there a chance this parameter is missing from the docs? or is it included and explained somewhere else?
Thanks

Briancbn, 2022-02-04:
I don't think it is inside the docs, @mathias-luedtke. You are welcome to contribute:)

mathias-luedtke, 2022-02-13:
`ISOLATION: "shell"` is not documented, as it is/was a beta feature and needs special configuration.

If you just want to access the builds from the outside, you can specify `BASEDIR`.