ros-industrial/industrial_ci #775: Builds are broken for ROS2 rolling

mathias-luedtke, 2022-02-13:
#774 will fix the configurations, but the repositories are not ready yet.

Briancbn, 2022-02-14:
Also the official rolling docker images are not available in jammy yet. I don't think this affect industrial_ci tho.

destogl, 2022-02-23:
Yesterday the binaries are released, but is seems that `rosdep` still has some issues...
[example on ros2_control](https://github.com/ros-controls/ros2_control/runs/5302992380?check_suite_focus=true#step:3:299)

Does something else should be updated too?

mathias-luedtke, 2022-02-23:
I believe that the mirror did not pick the changes up yet.

Briancbn, 2022-02-23:
Hi @destogl, I have the same issue using the ubuntu jammy docker image, install and setup rosdep through command line. So I think it's more of a rosdep issue atm.

vatanaksoytezer, 2022-02-24:
I'm only getting GPG errors in `apt update` when I include `testing` repositories. `main` repositories worked fine for me except that rosdep is acting funky and not installing all the dependencies. @mathias-luedtke let me know if I can be helpful in anyway, and I will gladly help. We want to migrate our CI to Rolling + Jammy as soon as possible. and love using industrial_ci.

mathias-luedtke, 2022-02-24:
> I'm only getting GPG errors in apt update when I include testing repositories. main repositories worked fine for me

It turned out that this a seccomp issue
```
docker run -it --rm --security-opt seccomp:unconfined ubuntu:jammy apt update
```
works

mathias-luedtke, 2022-02-24:
Not sure whats going on with rosdep:
```
rosdep resolve ament_cmake_python

ERROR: No definition of [ament_cmake_python] for OS version [Jammy Jellyfish]

No definition of [ament_cmake_python] for OS version [Jammy Jellyfish]
        rosdep key : ament_cmake_python
        OS name    : ubuntu
        OS version : Jammy Jellyfish
        Data:
_is_ros: true
                debian:
                  bullseye:
                    apt:
                      packages:
                      - ros-rolling-ament-cmake-python
                osx:
                  homebrew:
                    packages:
                    - ros/rolling/ament_cmake
                rhel:
                  '8':
                    dnf:
                      packages:
                      - ros-rolling-ament-cmake-python
                ubuntu:
                  jammy:
                    apt:
                      packages:
                      - ros-rolling-ament-cmake-python
```

Briancbn, 2022-02-24:
Strange thing is in Focal rolling error `OS_VERSION` seems to resolve as `focal` instead of `Focal F.....`. https://github.com/ros-tooling/action-ros-ci/issues/722#issuecomment-1040337895

mathias-luedtke, 2022-02-24:
Seems to a result of https://github.com/ros-infrastructure/rospkg/issues/241

nuclearsandwich, 2022-02-25:
rospkg 1.4.0 is uploading now and contains a fix for this issue among other changes.

mathias-luedtke, 2022-02-25:
rolling build is passing again, thank you very much @nuclearsandwich!

closed 2022-02-25 by mathias-luedtke