ros-industrial/industrial_ci #778: Skip build step for catkin_lint

BarisYazici, 2022-03-03:
Is it also possible to skip the build step for some jobs. I guess it is possible to workaround with the caching. But is there any feature to skip build for catkin_lint or clang-tidy lint in a similar way we have for testing(NO_TEST_BUILD)

BarisYazici, 2022-03-06:
I guess caching is not working for cpp-linting and catkin_linting stages. In both cases it can't find the build devel and logs directories.

mathias-luedtke, 2022-03-06:
catkin_lint and clang-tidy are part of the normal source builds (no need to configure multiple jobs!).
For catkin_lint you can skip the build itself with `NO_TEST_BUILD`.
For clang-tidy you have to build the code. Caching should work as well.

BarisYazici, 2022-03-07:
Thanks! There might be a potential bug in the caching for clang-tidy in gitlab. But I also didn't dig deeper, there might also be problem in my setup.

katettudelft, 2023-10-31:
Forgive the thread necromancy, but I hope someone can clarify for me: what exactly does NO_TEST_BUILD control?

Does it mean 'build the target_ws, but don't run tests', or does it mean 'don't build target_ws at all'? Does it have any effect on upstream_ws?