ros-industrial/industrial_ci #782: ROS2 Rolling installation is broken

AndyZe, 2022-04-29:
See the CI log.

```
  $ sudo apt-get -qq install -y --no-upgrade --no-install-recommends ros-rolling-ros-environment | grep -E 'Setting up' 
  E: Unable to locate package ros-rolling-ros-environment
  'sudo apt-get -qq install -y --no-upgrade --no-install-recommends ros-rolling-ros-environment | grep -E 'Setting up' ' returned with 100
```

tylerjw, 2022-04-29:
This doesn't look like a problem with industrial_ci, maybe post it on a OR repo for that package so someone who could make a change to fix this will see it?

AndyZe, 2022-05-01:
Per [comment from an Open Robotics staff member](https://github.com/ros/ros_environment/issues/32) on the `ros_environment` repo, the `ros-environment` debian is available for Rolling. So it seems like something in `industrial_ci` is causing the issue. Got any hints to debug?

This is holding up MoveIt2 at the moment  :(

mathias-luedtke, 2022-05-01:
There seems to be an issue with the key setup:

> $ sudo apt-get update -qq
W: http://archive.ubuntu.com/ubuntu/dists/jammy/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2012-cdimage.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://archive.ubuntu.com/ubuntu/dists/jammy/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2018-archive.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: http://archive.ubuntu.com/ubuntu jammy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://archive.ubuntu.com/ubuntu jammy InRelease' is not signed.
W: http://security.ubuntu.com/ubuntu/dists/jammy-security/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2012-cdimage.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://security.ubuntu.com/ubuntu/dists/jammy-security/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2018-archive.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: http://security.ubuntu.com/ubuntu jammy-security InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://security.ubuntu.com/ubuntu jammy-security InRelease' is not signed.
W: http://archive.ubuntu.com/ubuntu/dists/jammy-updates/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2012-cdimage.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://archive.ubuntu.com/ubuntu/dists/jammy-updates/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2018-archive.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: http://archive.ubuntu.com/ubuntu jammy-updates InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://archive.ubuntu.com/ubuntu jammy-updates InRelease' is not signed.
W: http://archive.ubuntu.com/ubuntu/dists/jammy-backports/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2012-cdimage.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://archive.ubuntu.com/ubuntu/dists/jammy-backports/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/ubuntu-keyring-2018-archive.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: http://archive.ubuntu.com/ubuntu jammy-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://archive.ubuntu.com/ubuntu jammy-backports InRelease' is not signed.
E: Problem executing scripts APT::Update::Post-Invoke 'rm -f /var/cache/apt/archives/*.deb /var/cache/apt/archives/partial/*.deb /var/cache/apt/*.bin || true'

mathias-luedtke, 2022-05-01:
> There seems to be an issue with the key setup:

apt update is not even working in a normal ubuntu:jammy image
(Try `docker run -it --rm ubuntu:jammy apt update`)

So I guess there is not much we can do.

AndyZe, 2022-05-01:
Thanks for looking into it.  :+1:

AndyZe, 2022-05-02:
@mathias-luedtke would you mind triggering CI on `master` branch again? Should be any easy way to close this issue.

My coworker said `docker run -it --rm ubuntu:jammy apt update` worked for him, so I am confused.

tylerjw, 2022-05-02:
@mathias-luedtke it worked for me when I tried running this yesterday and it just worked when I tried now. I think this issue is probably resolved.

closed 2022-05-02 by AndyZe

mathias-luedtke, 2022-05-02:
I had to reinstall docker and now it works properly..