ros-industrial/industrial_ci #786: Could not determine ref type of version: remote: Repository not found (private github repository during "setup_upstream_workspace" step)

Minipada, 2022-07-04:
Hi,
I am basing my CI on a private container image which contains a .netrc file allowing me to git clone using https. I am using a github hosted runner.

So if i enter my container and execute
```bash
git clone https://github.com/ORG/REPO
```
It will work. But when running from the CI, I get this:
[image]

I also tried to enable netrc from outside the container without success.
I'm not sure where to continue from that.

Thanks

mathias-luedtke, 2022-07-04:
How are you configuring UPSTREAM_WORKSPACE? It looks like you are not setting the version..

Minipada, 2022-07-04:
I have a string defined: packages.repos

mathias-luedtke, 2022-07-04:
> It looks like you are not setting the version..

No, it just cannot connect.to the remote.

Have you setup a PAT?
See https://github.com/ros-industrial/industrial_ci/issues/574#issuecomment-745306081

mathias-luedtke, 2022-07-04:
> which contains a .netrc 

I guess that this contains you PAT..
Where did you put it?

Minipada, 2022-07-04:
I just tried passing it like in the comment you mentioned and it does not make a difference. My current token is stored in ~/.netrc:
```
machine github.com login USER password ghp_...
```

mathias-luedtke, 2022-07-04:
> If it's the docker image default user, then it should find it

Yes, the default user.
But I am not sure, if a have ever tested it with a non-root user.

Does is work locally with vcs abd that .netrc?

mathias-luedtke, 2022-07-04:
Since you already have a custom docker image, you could try [ISOLATION: shell](https://github.com/ros-industrial/industrial_ci/blob/master/.github/workflows/main.yml#L155-L172) as well.

Minipada, 2022-07-04:
> Does is work locally with vcs and that .netrc?

Yes

Minipada, 2022-07-04:
With ISOLATION: shell, i"m getting the same error. It fails on the private repo

mathias-luedtke, 2022-07-04:
Hmm, tricky.

Please run a bash in your docker image and try `vcs import `manually.
If this works, clone industrial_ci and your repo, `export ISOLATION=docker` and run [ci.sh](https://github.com/ros-industrial/industrial_ci/blob/master/ci.sh) directly from your repo folder.

Minipada, 2022-07-04:
I can't run ISOLATION=shell on my pc, i'm on arch... so there is no apt etc...
Otherwise, without it, it runs properly

Minipada, 2022-07-04:
A colleague tried on his Ubuntu 20. With ISOLATION=shell and CI=true and it worked.
I'm lost on this one

Minipada, 2022-07-04:
Ok, we solved it. The error was on my side...
There was a permission step in the action that somehow only triggered issue when fetching private repos.

Sorry for the trouble and thanks for the support

closed 2022-07-04 by Minipada