ros-industrial/industrial_ci #787: Unable to get ROS Noetic APT packages

fgrcar, 2022-07-21:
Hi,

I am using Industrial CI for the Bitbucket Pipelines, version for ROS Noetic and it worked great until yesterday when there appeared problems with getting some of the dependencies over the rosdep like:
`executing command [apt-get install -y -qq ros-noetic-moveit]
E: Unable to locate package ros-noetic-moveit
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y -qq ros-noetic-moveit] failed`

If I update the APT source list and set up the keys like described in the [Noetic installation guide](http://wiki.ros.org/noetic/Installation/Ubuntu), then the packages are normally accessible.

Since the CI stopped working without any significant change on my side and it's therefore blocking my production, I would be glad if anyone has an idea what exactly is going on here?

Best regards,
fgrcar

gavanderhoorn, 2022-07-21:
The ROS buildfarm appears to be experiencing some problems the past few days. Specifically with MoveIt packages on Noetic.

See [status_page/ros_noetic_default.html?q=REGRESSION](http://repositories.ros.org/status_page/ros_noetic_default.html?q=REGRESSION).

I don't believe this is caused by anything here.

fgrcar, 2022-07-21:
Ok. Thank you for the answer. I hope it will be resolved soon.

mathias-luedtke, 2022-07-23:
> If I update the APT source list and set up the keys like described in the [Noetic installation guide](http://wiki.ros.org/noetic/Installation/Ubuntu), then the packages are normally accessible.

industrial_ci uses the `testing` repositories by default (same as for the ROS buildfarm).
Temporarily you could use `ROS_REPO=main`. Normally I just configure one job for each of them.

Our noetic tests are still passing, so there is not much that we can do from our side.
For now I will leave this issue open, because other users might experience the same.

Please close it as soon as it is resolved :)

fgrcar, 2022-08-01:
Looks like the problems with buildfarm have been resolved. Closing the issue.

closed 2022-08-01 by fgrcar