ros-industrial/industrial_ci #792: GitHub Actions: Deprecating save-state and set-output commands

fmessmer, 2022-10-12:
I'm not exactly sure what action needs to be taken, but as of today I'm starting to see the following deprecation notice in my CI runs:
```
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
```
see https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

and I see `set-output` being used here: https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/folding/github_actions.sh#L29

fmessmer, 2022-10-12:
is it just adding around it? https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#echoing-command-outputs
```
echo "::echo::on"
echo "::set-output name=$1::$2"
echo "::echo::on"
```

mathias-luedtke, 2022-10-12:
@fmessmer: Thanks for reporting!

If I understand it correctly,

https://github.com/ros-industrial/industrial_ci/blob/d6643d77e140a574baa0180c7c2717c37d9a0cbd/industrial_ci/src/folding/github_actions.sh#L29
Needs to be replaced by
```bash
    echo "$1=$2" >> "$GITHUB_OUTPUT"
```
(see https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#setting-an-output-parameter)

@rhaschke: Please have a look as well!

mathias-luedtke, 2022-10-13:
@fmessmer: please check if the warning is gone now :)

closed 2022-10-13 by mathias-luedtke

fmessmer, 2022-10-14:
thanks @mathias-luedtke 
thanks @rhaschke 

for fixing this so quickly - the deprecation warnings are gone :+1:

peci1, 2022-12-29:
I'm still [seeing these warnings](https://github.com/ctu-vras/ros-utils/actions/runs/3800916527). Is there anything I need to do to "update" industrial_ci for my repo? The workflow file is here: https://github.com/ctu-vras/ros-utils/blob/master/.github/workflows/ci.yaml . It seems to pull ici@master, which I'd guess should be sufficient.

rhaschke, 2022-12-30:
The first warning also states:
`Please update the following actions to use Node.js 16: actions/checkout@v2, actions/cache@v2`

Update them to version `v3`!

peci1, 2022-12-30:
Thanks @rhaschke! That was it, so it was a false "accusation" of ICI :)

PR #793 (2022-10-13): Migrate GitHub action to Environment file
set-output is deprecated

closes #792