ros-industrial/industrial_ci #796: Allow output of clang-tidy fixes file

EzraBrooks, 2022-10-14:
I'd like to be able to keep the clang-tidy fixes file generated during an industrial_ci run, so I can load it into other tools that generate reports - or just export it and use it to run fixes.

rhaschke, 2022-10-15:
Have a look at [this example](https://github.com/ros-planning/moveit/blob/afa202ec2c7ca312aca10029daa46a91be854543/.github/workflows/ci.yaml#L145-L150). Here, the applied changes are provided as a build artifact in GHA.

EzraBrooks, 2022-10-17:
That link doesn't work for me for some reason, but I was able to see the same thing on the main branch. Is that uploading a file called `clang-tidy`? Is that formatted the same way as a fixes file, or is it a different format of report generated by ICI?

rhaschke, 2022-10-17:
No, the provided file is directly a git patch file.

EzraBrooks, 2022-10-17:
Okay, well then my feature request still stands 😄

It would be nice to have the actual content of the fixes file for tools that accept it, for instance [this Action](https://github.com/marketplace/actions/clang-tidy-action) that can annotate clang-tidy fixes directly onto PR diffs.

rhaschke, 2022-10-17:
I see.
@mathias-luedtke, what do you suggest to transfer a file generated within the ICI docker container to the calling GHA process?

mathias-luedtke, 2022-10-18:
It should be possible to to do that with the current set of features.
* set CLANG_TIDY_ARGS to export the fixes file
* expose the result (if not already accessible): use shell isolation, set BASEDIR or DOCKER_RUN_OPTS

We could think of always exporting the file, if the default location is reasonable.

mathias-luedtke, 2022-10-18:
@EzraBrooks: Do you know if multiple fixes files can be merged (or at least appended). And we might have a problem with the different paths (inside workspace and repository)..

EzraBrooks, 2022-10-18:
I already tried adding a new -export-files argument via that environment variable. Unfortunately it can only be passed once, so the fact that industrial_ci's internals already pass it means adding it to that variable crashes the tests.

EzraBrooks, 2022-10-18:
Unsure about merging them. I was able to produce reasonable output in my project's workspace using `run-clang-tidy` on the workspace root, instead of `clang-tidy` on each package, though. But since I was doing that outside the context of the industrial_ci container I didn't have any of the dependencies I needed for its analysis, which is why I'd like the feature in ICI.

rhaschke, 2022-10-19:
I will have a look. Being able to provide feedback in github would be great!

closed 2025-12-11 by mathias-luedtke