ros-industrial/industrial_ci #802: How to build specific set of packages in upstream and target that I depend on

sisaha9, 2022-12-28:
I have 2 workspaces: The target workspace and an upstream workspace. I depend on only some of the packages in the upstream workspace. In turn I have several build targets in my target workspace and only want to test some of them in CI

How do I only build some packages from the target workspace and only build those packages from the upstream workspace that the target workspace depends upon?

yyagin, 2023-11-22:
Hi, have you found a solution to this? If so, could you please share it here? I'm facing the same problem.

sisaha9, 2023-11-22:
I haven't. I ended up just doing a completely custom solution to have better control

mathias-luedtke, 2023-11-24:
Have you tried deleting the folders that you do not need?

https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#workspace-definition

> directory path prefixed with - to remove the directory, as a path relative to either the source space or the target repository