ros-industrial/industrial_ci #805: pylint: Job fails even when there's no error/failure

130s, 2023-01-04:
# Issue

An ICI-powered job with `PYLINT_CHECK: "true"` fails even when `pylint` returns no error.

E.g. in https://github.com/ros-industrial/industrial_ci/pull/804#issuecomment-1370958300, where `PYLINT_ARG: "--errors-only"` is set (i.e. commit author is only interested in error) and `pylint` returned code 28 (no error included, `pylint` doesn't report it as a failure, I think). But ICI-powered CI job still failed.

# Condition of Satisfaction
- [ ] Confirmed the reported condition is a problem indeed.
- [ ] Agreed solution approach.

130s, 2023-01-04:
@fmessmer As an original author of `pylint` feature https://github.com/ros-industrial/industrial_ci/pull/612 (thanks!), do you have an insight? Do you have a workaround?

Dumb workaround may be go fix everything so that `pylint` would return code 0, but that's a wrong way to go if `pylint` marks many lines for my codebase.

fmessmer, 2023-01-05:
@130s see https://github.com/ros-industrial/industrial_ci/pull/804#pullrequestreview-1237391128
:wink:

closed 2023-01-18 by mathias-luedtke