ros-industrial/industrial_ci #807: Trouble cloning multiple private repositories from .repos file

TrevorGibson-SR, 2023-01-10:
Hi all,

I've been trying to get a GitHub Actions CI build to work with cloning more than one private repository, but have run into multiple issues and can't seem to resolve them.

### Overview
The current Action I have set up works fine specifically when only 1 private GitHub repository is listed in our `.repos` file. It breaks when either a second private GitHub repo is listed (issue 1), or any private BitBucket repositories are listed (issue 2).

Some additional information:
- Any number of public repositories are able to cloned successfully without issue
- Cloning multiple private repos from a local development machine (using `vcs import`) works fine, regardless of the source of those repos (GitHub or BitBucket)

Here is a snippet of the .yml action showing the steps being run (placeholders in place of actual repo URLs):

```
reusable_industrial_ci_with_cache:
    name: ${{ inputs.ros_distro }} ${{ inputs.ros_repo }} ${{ inputs.os_code_name }}
    runs-on: ubuntu-latest
    env:
      CCACHE_DIR: ${{ github.workspace }}/${{ inputs.ccache_dir }}
      BASEDIR: ${{ github.workspace }}/${{ inputs.basedir }}
      CACHE_PREFIX: ${{ inputs.ros_distro }}-${{ inputs.os_code_name }}-${{ inputs.ros_repo }}-${{ github.job }}
    strategy:
      fail-fast: false
    steps:
      - name: Checkout ${{ inputs.ref }} when build is not scheduled
        if: ${{ github.event_name != 'schedule' }}
        uses: actions/checkout@v3
      - name: Checkout ${{ inputs.ref }} on scheduled build
        if: ${{ github.event_name == 'schedule' }}
        uses: actions/checkout@v3
        with:
          ref: ${{ inputs.ref_for_scheduled_build }}
      - name: Setup SSH keys for private repositories 1
        uses: webfactory/ssh-agent@v0.7.0
        with:
          ssh-private-key: |
            ${{ secrets.KEY1 }}
            ${{ secrets.KEY2 }}
            ${{ secrets.KEY3 }}
      - name: Prepare git and ssh config for build context
        run: |
          mkdir root-config
          cp -r ~/.gitconfig ~/.ssh root-config/
      - name: cache target_ws
        if: ${{ ! matrix.env.CCOV }}
        uses: pat-s/always-upload-cache@v2.1.5
        with:
          path: ${{ env.BASEDIR }}/target_ws
          key: target_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('**/CMakeLists.txt', '**/package.xml') }}-${{ github.run_id }}
          restore-keys: |
            target_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('**/CMakeLists.txt', '**/package.xml') }}
      - name: cache ccache
        uses: pat-s/always-upload-cache@v2.1.5
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ env.CACHE_PREFIX }}-${{ github.sha }}-${{ github.run_id }}
          restore-keys: |
            ccache-${{ env.CACHE_PREFIX }}-${{ github.sha }}
            ccache-${{ env.CACHE_PREFIX }}
      - uses: 'ros-industrial/industrial_ci@master'
        env:
          UPSTREAM_WORKSPACE: ${{ inputs.upstream_workspace }}
          ROS_DISTRO: ${{ inputs.ros_distro }}
          ROS_REPO: ${{ inputs.ros_repo }}
          OS_CODE_NAME: ${{ inputs.os_code_name }}
          BEFORE_INSTALL_UPSTREAM_DEPENDENCIES: ${{ inputs.before_install_upstream_dependencies }}
      - name: prepare target_ws for cache
        if: ${{ always() && ! matrix.env.CCOV }}
        run: |
          du -sh ${{ env.BASEDIR }}/target_ws
[... 107 lines cut ...]
  Setting up python3-packaging (21.3-1) ...
  Setting up python3-pluggy (0.13.0-7.1) ...
  Setting up python3-notify2 (0.3-4) ...
  Setting up xml-core (0.18+nmu1) ...
  Setting up libxml2:amd64 (2.9.13+dfsg-1ubuntu0.2) ...
  Setting up libarchive13:amd64 (3.6.0-1ubuntu1) ...
  Setting up python3-pytest (6.2.5-1ubuntu2) ...
  Setting up python3-colcon-core (0.11.0-1) ...
  Setting up python3-colcon-notification (0.2.14-1) ...
  Setting up python3-colcon-pkg-config (0.1.0-1) ...
  Setting up python3-colcon-zsh (0.4.0-1) ...
  Setting up python3-colcon-library-path (0.2.1-1) ...
  Setting up cmake (3.22.1-1ubuntu1.22.04.1) ...
  Setting up python3-colcon-metadata (0.2.5-1) ...
  Setting up python3-colcon-python-setup-py (0.2.7-2) ...
  Setting up python3-colcon-package-information (0.3.3-1) ...
  Setting up python3-colcon-output (0.2.12-2) ...
  Setting up python3-colcon-package-selection (0.2.10-2) ...
  Setting up python3-colcon-defaults (0.2.6-1) ...
```

### Summary
Any help with either of these issues would be greatly appreciated. Ideally I would like to use the BitBucket private repo as it works better with the larger project architecture. However, cloning multiple private GitHub repositories is also a must for another project, so resolving that one alone will be helpful as well.

Thanks in advance

TrevorGibson-SR, 2023-01-24:
### Update for Issue 2

I was able to solve the issue of cloning a single non-GitHub private repo in the Docker container by adding that repository's server to the `known_hosts` file. I believe this is what was causing the build process to hang as it was waiting for user input to add the server to the list of known hosts.

This was fixed by issuing the following command immediately after the `ssh-agent` action in my GitHub workflow:

`ssh-keyscan -H bitbucket.org >> ~/.ssh/known_hosts`

After investigating the contents of the `known_hosts` file, it appears that github.com had already been added (either by the workflow managing agent, or by the `ssh-agent` action), which is why there weren't any issues cloning GitHub repositories.

This solution should work for repositories hosted on any non-GitHub servers... just replace `bitbucket.org` with the repo server's domain.

TrevorGibson-SR, 2023-02-20:
### Update for Issue 2

The root cause is due to the configuration created by webfactory/ssh-agent with the [deploy key mapping](https://github.com/webfactory/ssh-agent#support-for-github-deploy-keys) feature. The custom .gitconfig settings are not copied into the docker automatically, therefore not allowing more than one private GitHub repo to be cloned.

This can be resolved by passing both the `~/.gitconfig` file and `~/.ssh` directory as mapped locations into the docker container. For an immediate fix, [use the `DOCKER_RUN_OPTS` parameter](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#pass-custom-variables-to-docker) to specify the locations to map in your Github Actions workflow yaml file.

A more permanent fix has been [submitted as a PR](https://github.com/ros-industrial/industrial_ci/pull/815) to incorporate mapping these locations as part of the standard codebase.

AndyZe, 2023-12-03:
I'm having some trouble with this as well, but I haven't caught up with @TrevorGibson-SR yet.

Basically I've followed the instructions in [index.rst](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst):

```
In "Add a variable" section, fill in the following text field/area.

    Key: SSH_PRIVATE_KEY
```

Except I have 4 private repositories, so I figured it _might_ work to add 4 such private keys (with the corresponding public keys in the corresponding 4 private repos).

`SSH_PRIVATE_KEY_1`, `SSH_PRIVATE_KEY_2`, ..., `SSH_PRIVATE_KEY_4`

But, that doesn't work. `industrial_ci` hangs forever when running in Github. Any simple solution to this?

mathias-luedtke, 2023-12-04:
@AndyZe: Just add something like this to your before_script in `.gitlab-ci.ym`l

https://github.com/ros-industrial/industrial_ci/blob/3ed9846c96ed1e0bb36193e8e250632eaac980d0/gitlab.sh#L36-L44

mathias-luedtke, 2023-12-04:
Just saw that you are talking about Github as well..
The SSH auto-setup is only implemented for Gitlab.
However, a similar trick should work with Github. As far as I can tell, that's what` webfactory/ssh-agent` is doing under the hood.

As @TrevorGibson-SR already pointed out: it is crucial to setup the known hosts properly.

mathias-luedtke, 2023-12-04:
Multiple deploy keys should work with` webfactory/ssh-agent `and #844

AndyZe, 2025-01-17:
It's approximately a year later and I finally had occasion to try this again with github. I got it working by adding the `SSH_PRIVATE_KEY` and `SSH_SERVER_HOSTKEYS` as described here:  https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#gitlab-ci-access-to-private-repositories

Then I added an action to check out each dep:

```  - uses: actions/checkout@v4 # dependency
        with:
          ssh-key: ${{ secrets.SSH_PRIVATE_KEY }}
          repository: my_company/my_repo
          path: my_repo
```

So it really wasn't bad. AFAICT you could close this issue now.