ros-industrial/industrial_ci #808: galactic prerelease regression

Nir-Az, 2023-01-15:
Hi, Galactic prerelease started failing with no reason 
[image]

Is that a known issue?
https://github.com/IntelRealSense/realsense-ros/actions/runs/3924821400/jobs/6709332697

```
Traceback (most recent call last):
2023-01-15T18:50:57.5926134Z   File "/tmp/ros_buildfarm/ros_buildfarm/scripts/devel/create_devel_task_generator.py", line 287, in resolve_names
2023-01-15T18:50:57.5926478Z     resolved_names = resolve_for_os(
2023-01-15T18:50:57.5926926Z   File "/usr/lib/python3/dist-packages/rosdep2/catkin_support.py", line 87, in resolve_for_os
2023-01-15T18:50:57.5927261Z     d = view.lookup(rosdep_key)
2023-01-15T18:50:57.5927651Z   File "/usr/lib/python3/dist-packages/rosdep2/lookup.py", line 206, in lookup
2023-01-15T18:50:57.5927969Z     return self.rosdep_defs[rosdep_name]
2023-01-15T18:50:57.5928265Z KeyError: 'ros_workspace'
2023-01-15T18:50:57.5928407Z 
2023-01-15T18:50:57.5928556Z During handling of the above exception, another exception occurred:
2023-01-15T18:50:57.5928751Z 
2023-01-15T18:50:57.5928857Z Traceback (most recent call last):
2023-01-15T18:50:57.5929194Z   File "/tmp/ros_buildfarm/scripts/devel/create_devel_task_generator.py", line 21, in <module>
2023-01-15T18:50:57.5929756Z     run_module('ros_buildfarm.scripts.devel.create_devel_task_generator', run_name='__main__')
2023-01-15T18:50:57.5930114Z   File "/usr/lib/python3.8/runpy.py", line 210, in run_module
2023-01-15T18:50:57.5930423Z     return _run_code(code, {}, init_globals, run_name, mod_spec)
2023-01-15T18:50:57.5930740Z   File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
2023-01-15T18:50:57.5930990Z     exec(code, run_globals)
2023-01-15T18:50:57.5931328Z   File "/tmp/ros_buildfarm/ros_buildfarm/scripts/devel/create_devel_task_generator.py", line 304, in <module>
2023-01-15T18:50:57.5931654Z     sys.exit(main())
2023-01-15T18:50:57.5931966Z   File "/tmp/ros_buildfarm/ros_buildfarm/scripts/devel/create_devel_task_generator.py", line 126, in main
2023-01-15T18:50:57.5932418Z     debian_pkg_names += resolve_names(['ros_workspace'], **context)
2023-01-15T18:50:57.5932812Z   File "/tmp/ros_buildfarm/ros_buildfarm/scripts/devel/create_devel_task_generator.py", line 290, in resolve_names
2023-01-15T18:50:57.5933137Z     raise RuntimeError(
2023-01-15T18:50:57.5933481Z RuntimeError: Could not resolve the rosdep key 'ros_workspace'
```

Nir-Az, 2023-01-18:
I see galactic fails at this repo tests too.
Does anyone plan to fix this build or EOL distro (galactic) cannot work and should be removed?

mathias-luedtke, 2023-01-18:
Pre-release tests do not work for EOL distros.
This is a limitation of ros_buildfarm.

As you can see in in your log:
> 2023-01-15T18:50:57.5908602Z # BEGIN SECTION: Run Dockerfile - generating devel tasks
2023-01-15T18:50:57.5909012Z reading in sources list data from /etc/ros/rosdep/sources.list.d
2023-01-15T18:50:57.5909585Z Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
2023-01-15T18:50:57.5910083Z Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
2023-01-15T18:50:57.5910575Z Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
2023-01-15T18:50:57.5911061Z Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
2023-01-15T18:50:57.5911838Z Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
2023-01-15T18:50:57.5912451Z Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
2023-01-15T18:50:57.5913064Z Skip end-of-life distro "ardent"
2023-01-15T18:50:57.5913467Z Skip end-of-life distro "bouncy"
2023-01-15T18:50:57.5913849Z Skip end-of-life distro "crystal"
2023-01-15T18:50:57.5914232Z Skip end-of-life distro "dashing"
2023-01-15T18:50:57.5914612Z Skip end-of-life distro "eloquent"
2023-01-15T18:50:57.5914920Z Add distro "foxy"
**2023-01-15T18:50:57.5915264Z Skip end-of-life distro "galactic"**
2023-01-15T18:50:57.5915641Z Skip end-of-life distro "groovy"
2023-01-15T18:50:57.5916083Z Add distro "humble"
2023-01-15T18:50:57.5917914Z Skip end-of-life distro "hydro"
2023-01-15T18:50:57.5918398Z Skip end-of-life distro "indigo"
2023-01-15T18:50:57.5918890Z Skip end-of-life distro "jade"
2023-01-15T18:50:57.5919275Z Skip end-of-life distro "kinetic"
2023-01-15T18:50:57.5919886Z Skip end-of-life distro "lunar"
2023-01-15T18:50:57.5920230Z Add distro "melodic"
2023-01-15T18:50:57.5920670Z Add distro "noetic"
2023-01-15T18:50:57.5920955Z Add distro "rolling"

So rosdep cannot resolve rosdep keys anymore..

Nir-Az, 2023-01-18:
Thanks for the info.
Does this means that it is not possible to `bloom-release` to galactic anymore?

mathias-luedtke, 2023-01-18:
> Does this means that it is not possible to bloom-release to galactic anymore?

At least not for regular releases..

PR #809 (2023-01-18): Notify that pre-release tests are not supported for EOL distros
closes #808 

@Nir-Az: Would you prefer that the jobs succeeds, but still prints the error?