ros-industrial/industrial_ci #821: Broken link [OS_NAME]

matthew-william-lock, 2023-03-25:
Looking at the detailed documentation, the link on the following line:
>OS_NAME (default: derived from OS_CODE_NAME): Possible options: {ubuntu, debian}. See [this section for the detail](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#optional-type-of-os-and-distribution).

does not take the user anywhere.

Furthermore, the link to the following line produces a 404
> Possible combination of OS_NAME and OS_CODE_NAME depend on available Docker images. See [ros-industrial/docker/ci](https://github.com/ros-industrial/docker/tree/master/ci).

mortenfyhn, 2023-11-04:
Ditto for the links on this line:

https://github.com/ros-industrial/industrial_ci/blob/9e8b4aaec69f7cade96eda5cfcb3e7da57b99c6e/doc/index.rst?plain=1#L89

mathias-luedtke, 2024-08-04:
Thanks for reporting those broken links!
I think I fixed/removed them in https://github.com/ros-industrial/industrial_ci/pull/874

closed 2024-08-04 by mathias-luedtke