ros-industrial/industrial_ci #822: prerelease test broken if custom CCACHE_DIR is used

rhaschke, 2023-03-30:
[This build](https://github.com/rhaschke/rviz/actions/runs/4564740083) using a custom `CCACHE_DIR=/home/runner/work/rviz/rviz/.ccache` fails with the following error:
```
# BEGIN SECTION: Run Dockerfile - build and install
# BEGIN SUBSECTION: build workspace in isolation and install
...
ccache: error: Failed to create directory /home/buildfarm/.ccache/tmp: Permission denied
```

The `CCACHE_DIR` is correctly passed to docker run by `industrial_ci`:

<tt>docker create --init --env-file /home/runner/work/_actions/ros-industrial/industrial_ci/master/industrial_ci/src/isolation/docker.env --rm -e TRAVIS -e OS_NAME -e OS_CODE_NAME -e OS_ARCH -e PRERELEASE_DOWNSTREAM_DEPTH -e PRERELEASE_REPONAME -e ROSDISTRO_INDEX_URL -e PRERELEASE_DISTRO -v /tmp/tmp.xNEHYg9fGl:/tmp/tmp.xNEHYg9fGl:rw -e WORKSPACE=/tmp/tmp.xNEHYg9fGl -v /run/docker.sock:/run/docker.sock:rw <strong>-v /home/runner/work/rviz/rviz/.ccache:/tmp/tmp.xNEHYg9fGl/home/.ccache:rw -e CCACHE_DIR=/tmp/tmp.xNEHYg9fGl/home/.ccache</strong> -v /home/runner/work/rviz/rviz:/home/runner/work/rviz/rviz:ro -e TARGET_REPO_PATH=/home/runner/work/rviz/rviz -v /home/runner/work/_actions/ros-industrial/industrial_ci/master/industrial_ci/src:/home/runner/work/_actions/ros-industrial/industrial_ci/master/industrial_ci/src:ro -e ICI_SRC_PATH=/home/runner/work/_actions/ros-industrial/industrial_ci/master/industrial_ci/src -v /home/runner/work:/home/runner/work:rw -e BASEDIR=/home/runner/work -v /home/runner/work/_temp/_runner_file_commands/set_output_f3392e2a-41ca-442c-8479-71cfa390e1c6:/home/runner/work/_temp/_runner_file_commands/set_output_f3392e2a-41ca-442c-8479-71cfa390e1c6:rw -e GITHUB_OUTPUT=/home/runner/work/_temp/_runner_file_commands/set_output_f3392e2a-41ca-442c-8479-71cfa390e1c6 -t --entrypoint  -w /home/runner/work/rviz/rviz ros:noetic-ros-core /bin/bash /home/runner/work/_actions/ros-industrial/industrial_ci/master/industrial_ci/src/run.sh ros_prerelease run_ros_prerelease
</tt>

This used to work [8 months ago](https://github.com/ros-visualization/rviz/actions/runs/2773462999).
I'm afraid that's a bug introduced by the upstream prerelease scripts, but I don't have time now to investigate :disappointed:

mathias-luedtke, 2024-08-04:
#870 uses the latest ros_buildfarm version, which _might_ have improved CCACHE_DIR support

rhaschke, 2024-08-05:
Unfortunately, the error still persists: https://github.com/ros-visualization/rviz/actions/runs/10246767919/job/28344566578