ros-industrial/industrial_ci #838: ROS Noetic Prelease failure

johnwason, 2023-07-07:
I am seeing a failure during a ROS Noetic Prerelease test of a cmake package:

```
Test results of underlay workspace
  
  Invoking: catkin_test_results /tmp/tmp.aWbcTaP5Te/ws/test_results --all
  Test results directory "/tmp/tmp.aWbcTaP5Te/ws/test_results" does not exist
```

The full workflow run can be found here:

https://github.com/robotraconteur/robotraconteur/actions/runs/5481982662/jobs/9986864176

These tests work properly for ROS Humble and Iron

The GitHub action job (run on branch ros-noetic):

```
build-ros-prerelease:
    runs-on: ubuntu-latest
    steps:
    - uses: actions/checkout@v3
    - name: ROS Distro
      run: echo "ROS_DISTRO=$(echo "$GITHUB_REF" | sed 's/^.*ros\(2\?\)-\([[:alnum:]]*\).*$/\2/')" >> $GITHUB_ENV
    - name: ros industrial-ci
      uses: ros-industrial/industrial_ci@master
      env:
        ROS_DISTRO: ${{ env.ROS_DISTRO }}
        ROS_REPO: main
        PRERELEASE: true
```

mathias-luedtke, 2024-08-04:
@johnwason: Sorry, did not see this issues before. #870 should fix that as well.

johnwason, 2024-08-10:
I am still seeing an error:

```
Test results of underlay workspace
  Invoking: catkin_test_results /tmp/tmp.amnW4dr9bq/ws/test_results --all
  Test results directory "/tmp/tmp.amnW4dr9bq/ws/test_results" does not exist
  'sudo -EH -u ci sh -c . /opt/ros/noetic/setup.sh && cd '/tmp/tmp.amnW4dr9bq' && exec ./prerelease.sh -y' returned with 1
'run_prerelease_script' returned with code '1' after 30 min 8 sec
```

https://github.com/johnwason/robotraconteur/actions/runs/10327274040/job/28592196144

My package uses cmake instead of catkin.