ros-industrial/industrial_ci #839: Enable usage of custom Clang-Tidy version

ardabbour, 2023-08-10:
It would be great to use a custom version of Clang-Tidy, especially given that clang-tidy-10, which is the default version of ubuntu:20.04, raises a `clang-diagnostic-error` on several `actionlib` headers. As a workaround, one can do something like this:

```yaml
variables:
  BEFORE_CLANG_TIDY_CHECKS : |
    if [ "$EUID" -ne 0 ] && command -v sudo > /dev/null; then
      sudo apt-get -qq install -y --no-upgrade --no-install-recommends clang-tidy-12
      sudo ln -sf $(which clang-tidy-12) $(which clang-tidy)
    else
      apt-get -qq install -y --no-upgrade --no-install-recommends clang-tidy-12
      ln -sf $(which clang-tidy-12) $(which clang-tidy)
    fi
```

But it would waste less CI minutes and overall be cleaner if there was a way to choose the Clang-Tidy version of interest, as is done with Clang-Format.

mathias-luedtke, 2023-08-12:
Why don't you use ADDITIONAL_DEBS?
Do other versions not setup set themselves as default?

ardabbour, 2023-08-12:
I believe the version used for setup changes depending on the base OS, whatever the default version is gets aliased to the `clang-tidy` command, which is then used for the checks. Even if the required version is installed via `ADDITIONAL_DEBS`, the `clang-tidy` command used later on will trigger the OS's default Clang-Tidy version. Also, installing the default version is a (minor) waste of CI time.